arduino/Arduino #6309: avrdude fails to upload sketches to Arduino Micro under linux

HackerInside0, 2017-05-21:
Instead it works perfectly under windows 10 without installing the drivers manually.

Here are the avrdude logs:
Under windows: 
https://gist.github.com/HackerInside0/eb3336ecd1b3f03e315930998d5b2065

Under linux ( user mode ):
https://gist.github.com/HackerInside0/f52a96abd3b2ac79dab7669be46ef644

Under linux ( as root ):
https://gist.github.com/HackerInside0/eb456815f097e4c1f8d38f6e788985ce

In this last case, sometime it freezes on : "avrdude: butterfly_recv(): programmer is not responding"

facchinm, 2017-05-22:
Hi @HackerInside0 ,
the problem is 99,999% related with `modemmanager`. Simply uninstall it from your Linux distribution (it is useless unless you use a 56k modem :slightly_smiling_face: ) and the problem should disappear.

HackerInside0, 2017-05-22:
It works now.
Thanks

closed 2017-05-22 by HackerInside0