arduino/Arduino #6312: Please add uMT library

GO01260, 2017-05-22:
uMT - a preemptive, real-time multitasker for the ARDUINO AVR/SAM boards

matthijskooijman, 2017-05-22:
This is a duplicate of #6300. According to that, the library was added and awaiting a proper tag by @GO01260. Looking at the new tag now, it looks ok, so I expect the library will be added automatically. If it doesn't appear within a few hours, best leave a comment on #6300.

closed 2017-05-22 by matthijskooijman

GO01260, 2017-05-22:
Thanks.
This is my first publication so I made a large series of mistakes... Sorry.

Any doc I can read which explains how to manage releases and how to create new versions of GitHub library?

I spent some time reading here and there but I did not get the concepts... (my fault).

Regards

cmaglie, 2017-05-24:
To create a new release you must change the version in the library.properties file, commit the change and make a new tag (just remember to create the new tag **after** having committed and pushed the changes to library.properties). Once done the new release will be published within an hour.

To remove a release you must remove the corresponding tag. The release will be actually removed from the index within a week (because our indexer runs a clean up job every week).