arduino/Arduino #6314: Arduino mega freezes if send "!!!" by serial

daiolix, 2017-05-23:
While debugging my progect have found that arduino mega 2560 freezes and holding serial port if i try to send "!!!" somewhere in code. Only if unplug and plug the board i can continue. Progect didn't download, board is empty.
I don't know is it bug, or "!!!" is fobbiden.
Have tested with code below, tried different speeds. Same with Serial.write
Now i can't test it with uno.

void setup() {
  Serial.begin(9600);
  Serial.print("!!!");
 }

void loop() { 
}

PaulStoffregen, 2017-05-23:
Are you using a genuine Arduino Mega?

This problem was fixed many years ago.

matthijskooijman, 2017-05-23:
This is a duplicate of #459. To fix it, you'll have to update the bootloader in the mega. The version shipped with the Arduino IDE should have the fix, so you can use the "Burn Bootloader" option of the IDE. You will need an ISP programmer (or another arduino using the ArduinoISP sketch), Google probably has pointers on how to set this up.

closed 2017-05-23 by matthijskooijman