arduino/Arduino #6315: [Library Manager] Please add micro-ecc

d00616, 2017-05-23:
URL: https://github.com/kmackay/micro-ecc by @kmackay

A small and fast ECDH and ECDSA implementation for 8-bit, 32-bit, and 64-bit processors.

The static version of micro-ecc (ie, where the curve was selected at compile-time) can be found in the "static" branch.

Features

- Resistant to known side-channel attacks.
- Written in C, with optional GCC inline assembly for AVR, ARM and Thumb platforms.
- Supports 8, 32, and 64-bit architectures.
- Small code size.
- No dynamic memory allocation.
- Support for 5 standard curves: secp160r1, secp192r1, secp224r1, secp256r1, and secp256k1.
- BSD 2-clause license.

NicoHood, 2017-05-23:
@d00616 secp256k1 is not marked as secure, so I am unsure if the use of this is recommended. Do you know if this was discussed somewhere? What about implementing Curve25519?
https://safecurves.cr.yp.to/

cmaglie, 2017-05-24:
Done!

closed 2017-05-24 by cmaglie