arduino/Arduino #6316: Comment highlighting

sullivan-wenger, 2017-05-23:
When I need to quickly switch between two blocks of code, I add in this construct to speed up switching:
//*
<block1>
/*/
<block2>
//*/
which comments out block 2 and leaves block 1 uncommented. Then, I just have to delete the slash at the start to simultaneously uncomment block 2, and comment block 1. However, if I comment out block 2, the IDE highlights both blocks as though they're not commented.

per1234, 2017-07-06:
Which version of the Arduino IDE are you using?

Which operating system?

I can't reproduce this on Arduino IDE 1.8.3 with Windows 7 64 bit:
[image]

sullivan-wenger, 2017-07-06:
My operating system is Ubuntu 16.04, 64 bit.
I can't find the version, but I know it's the one I got with Ubuntu's `sudo apt install arduino`. I've found something on it that says 1.0.5, but I don't know if that's what it is.

Also, rereading my bug report, I'm realizing I don't like how I described the issue. Let me try again:

This code:
```
/*
block1 //highlighted as a comment (correct)
/*/
block2 //not highlighted as a comment (correct)
// */
```
is just fine. But this code:
```
//*
block1 //not highlighted as a comment (correct)
/*/
block2 //not highlighted as a comment (INCORRECT- should be a comment)
// */
```
has a problem.

per1234, 2017-07-07:
Unfortunately `sudo apt install arduino` installs a very outdated and possibly non-standard version of the Arduino IDE (see https://github.com/arduino/Arduino/pull/2703).

As you can see from the screenshot I posted above, comment highlighting is working as you would like in Arduino IDE 1.8.3. Try installing the latest version downloaded from https://www.arduino.cc/en/Main/Software and let me know if it solves this issue for you. I think you'll find there has been a lot of work done on the IDE since 1.0.5!

per1234, 2017-08-03:
@sullivan-wenger did you try Arduino IDE 1.8.3 yet? If so, did it solve the issue?

per1234, 2017-08-22:
Closing as invalid due to lack of a response.

closed 2017-08-22 by per1234