arduino/Arduino #6321: Arduino Nano Eagle file broken

KateElizabeth112, 2017-05-25:
The Eagle file for the Arduino Nano schematic currently available is broken. I get the following error message "Error: line 140, column 25: Unexpected ''."

I believe there has been a fix from a previous thread, but the website has not been updated. Happy to apply the fix myself.

michaellass, 2017-05-27:
More or less connected to #6039 which was reopened now. There seems to exist a fix for the files which hasn't made it to the websites yet.

per1234, 2017-07-06:
Thanks for pointing that out michaellass! Closing as duplicate of https://github.com/arduino/Arduino/issues/6039 since the issue is already being tracked there and @KateElizabeth112 has already commented on that issue.

closed 2017-07-06 by per1234