arduino/Arduino #6322: IDE Failed to Open Arduino File/DC Motors

camandbass, 2017-05-26:
Hello Friends,

        I am using an Arduino Mega2560 r3, an L293D IC and a power supply module to run a DC motor. As instructed by the "most complete starter kit" eBook, I looked to my code folder for lesson 29, DC motors, and found an arduino file, as anticipated. 

        When i open the file with Arduino IDE 1.8.2, I get a error message saying Arduino IDE Failed to open, so I downloaded Developer Virtuoso, and I get the code below in text.

        When I upload this sketch, my DC motor spins anti-clockwise, pauses, repeats (also anti-clockwise) 4 more times, then there's a bigger delay, then it spins at a constant speed (still anti-clockwise) for a few seconds, then about a 10 second pause, repeat.

I can tell the code, although i didn't write it, would tell the motor to switch directions back and forth for a bit, to speed up and slow down, than show varying speeds in descending order.

Yet my DC motor shows no variety in speed or direction.

I know it's wired correctly, many different tripple checks, it is a match for what the lesson says and depicts, and what my training with IC's thinks it should be

My question is for what reason would Arduino IDE 1.8.2 not be able to open arduino files provided in the "most complete starter kit" when they could be accessed by Developer Virtuoso; and/or does the code below achieve what in its own footnotes claim to.

None of the code I downloaded from Elegoo in "arduino file" format can be opened by Arduino IDE 1.8.2, although it works great uploading sketches to my arduino mega; I've even reinstalled to no avail.

Basically  if this code below is valid, theres something wrong with my motor

//www.elegoo.com
//2016.12.12

/************************
Exercise the motor using
the L293D chip
************************/

#define ENABLE 5
#define DIRA 3
#define DIRB 4

int i;
 
void setup() {
  //---set pin direction
  pinMode(ENABLE,OUTPUT);
  pinMode(DIRA,OUTPUT);
  pinMode(DIRB,OUTPUT);
  Serial.begin(9600);
}

void loop() {
  //---back and forth example
    Serial.println("One way, then reverse");
    digitalWrite(ENABLE,HIGH); // enable on
    for (i=0;i<5;i++) {
    digitalWrite(DIRA,HIGH); //one way
    digitalWrite(DIRB,LOW);
    delay(500);
    digitalWrite(DIRA,LOW);  //reverse
    digitalWrite(DIRB,HIGH);
    delay(500);
  }
  digitalWrite(ENABLE,LOW); // disable
  delay(2000);

  Serial.println("fast Slow example");
  //---fast/slow stop example
  digitalWrite(ENABLE,HIGH); //enable on
  digitalWrite(DIRA,HIGH); //one way
  digitalWrite(DIRB,LOW);
  delay(3000);
  digitalWrite(ENABLE,LOW); //slow stop
  delay(1000);
  digitalWrite(ENABLE,HIGH); //enable on
  digitalWrite(DIRA,LOW); //one way
  digitalWrite(DIRB,HIGH);
  delay(3000);
  digitalWrite(DIRA,LOW); //fast stop
  delay(2000);

  Serial.println("PWM full then slow");
  //---PWM example, full speed then slow
  analogWrite(ENABLE,255); //enable on
  digitalWrite(DIRA,HIGH); //one way
  digitalWrite(DIRB,LOW);
  delay(2000);
  analogWrite(ENABLE,180); //half speed
  delay(2000);
  analogWrite(ENABLE,128); //half speed
  delay(2000);
  analogWrite(ENABLE,50); //half speed
  delay(2000);
  analogWrite(ENABLE,128); //half speed
  delay(2000);
  analogWrite(ENABLE,180); //half speed
  delay(2000);
  analogWrite(ENABLE,255); //half speed
  delay(2000);
  digitalWrite(ENABLE,LOW); //all done
  delay(10000);
}

Ultimate project is to use a motion sensor to turn on a fan for about a minute, then the fan turns off until prompted again.

Of course the blades are oriented to push air forwards when spinning clockwise

Thanks much if you suffer through reading this all

-Chris

camandbass, 2017-05-26:
whats with the scrolling text box at the top

camandbass, 2017-05-26:
also if i just switch my poles on the DC motor it will spin the correct way to make the fan work, but it still runs the code incorrectly with no change in speed or direction

per1234, 2017-05-26:
@camandbass's forum post for this issue:
http://forum.arduino.cc/index.php?topic=479714

facchinm, 2017-05-30:
Hi @camandbass ,
the sketch is not the problem here; thanks @per1234 for pointing out the forum post.
This is probably related with https://github.com/arduino/arduino-builder/issues/212 or another unicode related issue on the builder.
I'c closing this issue so we can continue debugging it in the right place.

closed 2017-05-30 by facchinm