arduino/Arduino #6325: ctrl-keyword

pfergiu, 2017-05-26:
Like in most of professional IDEs, it could be great to add a new feature that brings you to the class or function declaration in code or library, only pressing control key and clicking in a highlighted keyword.

lmihalkovic, 2018-04-30:
I think it would be nice if that was also extended to includes, giving the ability to view them in RO mode. i saw that 1.9.0 offers a nice workaround there, with code completion, but looking at a lib code is full of insights when starting

lmihalkovic, 2018-04-30:
It is possible to crudely extend the patch Ricardo created in __SketchTextArea__ to make it identify include references and fire custom URL protocol (here I called it 'aide' for 'arduino ide') reference containing the data to be displayed (the server name is 'absolute' or 'relative'). With a dummy protocol handler, the following will be displayed:

[image]

[image]

```
    Pattern regex = Pattern.compile("([\"<])([\\w./]+)[\">]");
    private Optional<HyperlinkEvent> getIncludeReference(Token t) {
      int begin = t.getOffset();      
      Token n1 = t.getNextToken();
      try {
        String line = getTextLine(t);
        Matcher m = regex.matcher(line);
        if (m.find()) {
          String desc = null;
          try {
            String svr = m.group(1).equals("<") ? "absolute" : "relative";
            URL url = new URL("aide", svr, m.group(2));
            HyperlinkEvent evt = new HyperlinkEvent(SketchTextArea.this, HyperlinkEvent.EventType.ACTIVATED, url, "blah blah");
            return Optional.ofNullable(evt);
          } catch (MalformedURLException mue) {
            desc = mue.getMessage();
          }
        }
      } catch (BadLocationException e) {
      }
      return Optional.empty();
    }
```