arduino/Arduino #6326: puzzled about digital and analog pin numbering Uno/Nano vs. Mega/Due

tofrnr, 2017-05-27:
hello,
I am puzzled about digital and analog pin numbering, and the documentation is not quite clear about that.
IIUC, some pin numbers >22 sometimes refer to analog pins (Uno) and sometimes to analog pins(Mega).

So if one wants to poll digital pins one by one, is it more safe (e.g. for the Mega) to poll
for (int i=2; i<50; i++)  int d =digitalRead(i);
or instead
for (int i=D2; i<D50; i++) int d =digitalRead(i);

and will it be better to read analog values by int numbers or Apin numbers one by one?

for (int i=2; i<16; i++)  int a =analogRead(i);
or instead
for (int i=A0; i<A16; i++) int a = analogRead(i);

or is it all the same?

facchinm, 2017-05-29:
Hi @tofrnr,
the pin numbering always tries to be consecutive for both analog and digital pins but it's not assured (there could be "holes" in the numbering sequence). When calling `digitalRead` or `analogRead` on these nonexisting pins you'll get `-1` as return so you can exclude them safely. Hope it answers your questions, feel free to reopen the issue if not

closed 2017-05-29 by facchinm

tofrnr, 2017-05-29:
thank you, about the "holes" and that  -1 return value it's now clear. 
But i's still not clear what the difference is about
digitalRead(24) vs. digitalRead(D24)
and
analogRead(24) vs. analogRead(A4) on an Uno
analogRead(10) vs. analogRead(A10) on a Mega
and what makes
digitalRead(A4)
instead

so actual the difference between "plain naumbers" and D... numbers and A... numbers is not clear and nowhere listed and explained in any documentation as far as I could see.

facchinm, 2017-05-29:
`D*` in not defined anywhere (maybe it is defined in 3rd party architectures but not in standard Arduino ones). To access digital pins you only need the plain number; instead analog pins real pin number is "shadowed" by the `A*` definition, and you can access them using both methods.
Real life example on a Mega: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/mega/pins_arduino.h#L68
calling `digitalRead(A0)` is the same as calling `digitalRead(54)` once you configure the pin as `INPUT`. 
`analogRead` instead is not applicable on non analog pins. For this reason, calling 
`analogRead(A0)` is the same as calling `analogRead(54)` (and it's also the same as calling `analogRead(0)` which represents the ADC channel and is still there for compatibility reasons). Instead, calling `analogRead(23)` (which is higher than the maximum analog pin number and lower than its shadow name) returns -1 again. 
The relevant piece of code is [here](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/wiring_analog.c#L48) 
I remember all these behaviours were documented somewher ein the website but I'mn unable to find them ATM, I'll post an update if I find them or bring them back online

tofrnr, 2017-05-29:
thank you - now it's already a bit more clear.
So also on an **Uno** or a **Nano** 
A0 == 54 for digitalRead and 
A0 == 54 == 0 for analogRead ?

facchinm, 2017-05-29:
Nope, you need to take a look at their respective variants:
UNO and NANO: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/variants/standard/pins_arduino.h#L56 .
If you want to make your sketches portable, always use `A*` way to access analog pins. 
About the return values, it is always signed, with values < 0 representing an error.

tofrnr, 2017-05-30:
thank you, actually the documentation is quite poor about all that and should be improved possibly.