arduino/Arduino #6328: Library structure: allow different version

aster94, 2017-05-29:
Hello, 
Maybe it is already possible and i don t know. If not it would be great to use different version of the same library. For examply i use the standard arduino board (i.e. Uno and nano) and also some attiny. When i change between them i need to modify the library to have it working. 
I think it would be useful

facchinm, 2017-05-29:
Hi @aster94 ,
it's totally possible, just put the hardware specific parts between `ifdefs` 
For example, if you need a function to run on the UNO only, you can enclose the code between
```
#ifdef ARDUINO_AVR_UNO
... (your code here) ....
#endif
```
of use `ARDUINO_ARCH_AVR` if you want to be more generic. The same applies to ATTINY architecture. 
If you have more questions on this topic please write on the forum, since it reaches a broader audience and these kind of questions are quite common :wink:

closed 2017-05-29 by facchinm

aster94, 2017-05-29:
Thank @facchinm!