arduino/Arduino #6332: Please consider adding a new repository specifically for managing addition (and removal) of libraries in the Board manager

feikname, 2017-05-29:
I feel like they're taking a substantial amount of the issues in this repository and getting out of hand for searching real issues related to the board manager. Or perhaps a `add-library-request` label could be added to them, so it'd be possible to blacklist it whilst doing searches.

matthijskooijman, 2017-05-30:
Sounds like a good plan! In addition to cluttering the issue list, they also cause a lot of noise among the email notififcation of issues.

In fact, perhaps the board manager config file / library list (which I assume exists) can be managed in git, so people can even submit PRs themselves for adding libraries?

feikname, 2017-08-31:
Apologies for pinging you guys here, but there has been no official reply since I created this issue 3 months ago.

@cmaglie @mastrolinux @agdl @facchinm

per1234, 2021-06-07:
All management of the Arduino Library Manager index is now to be done in a dedicated repository:
https://github.com/arduino/library-registry

I will be transferring and redirecting all incoming issues from this repository over there, getting this repo one step closer to being dedicated to the Arduino IDE.

I created a pinned issue to bring more awareness to the change: https://github.com/arduino/Arduino/issues/11588

closed 2021-06-07 by per1234