arduino/Arduino #6339: Can't uplod : Programmer is not responding

Zanatoos, 2017-05-31:
Hello everyone,

I'm coming to you because i have a problem that i've been trying to solve for days and i'm stuck...
So i wanted to modify the bootloader of my Arduino UNO by adding some functions of my own. And to do that, i had to modify the fuse settings of my board to get the maximum size (HFuse : D8, LFuse : FF, EFuse : FD). 
So i used another Arduino UNO to burn the bootloader and everything worked. I use the LED_START_FLASHES variable to check if the current bootloader is the good one. 

But then, it's impossible for me to upload an sketch on this board, i get the following output :+1: 
`avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : COM10
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : ATmega328P
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
         Hardware Version: 4728160
         Firmware Version: 0.4598883
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
         Vtarget         : 420030.5 V
         Varef           : 198234348.8 V
         Oscillator      : 1.373 Hz
         SCK period      : 7006770.9 us

avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_recv(): programmer is not responding
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_recv(): programmer is not responding

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
avrdude done.  Thank you.`

But if i put the original bootloader, everything works perfectly.

facchinm, 2017-05-31:
Hi @Zanatoos ,
which bootloader are you using as base? Starting from an up-to-date [optiboot](https://github.com/Optiboot/optiboot) is the best choice right now. Your setup looks ok and the fuses too, there seem to be synchronisation/baud rate problems which lead to the spurious values like `Hardware Version: 4728160` . If switching bootloader does not solve the issue consider posting on the forum to get a broader audience :slightly_smiling_face:

closed 2017-05-31 by facchinm

Zanatoos, 2017-05-31:
Hi Facchinm, 
i use the bootloader that came when i installed the Arduino IDE, in the bootloaders folder. I just modified, the makefile and optiboot.c, then I opened a terminal and did : 
`make clean`
then
`make atmega328 optiboot.c`

And after that i plug my other Arduino UNO, upload the example ArduinoISP, put the AVR ISP Shield with the Atmega chip i want to burn and then i burn the bootloader. 
Then I put back the chip on the board and when i plug it in, i count the number of flashes to check if it's the good bootloader.

facchinm, 2017-05-31:
That's fine, I just wanted to point out that the optiboot sources inside the IDE are at least 4 years old and in the meantime the compiler evolved a lot, so recompiling the original bootloader could give unexpected results (as you are experiencing), so it's better to start from the updated sources you can find here https://github.com/Optiboot/optiboot

Zanatoos, 2017-06-01:
Actually you were right, i tried to base my code on this uptodate optiboot and i've been able to upload a sketch with this bootloader with just setting my fuse settings. But when i add my functions, i had to change the makefile so it compiles all my files and not only optiboot.c. 
But then i have the error : 
"C:/winavr-20100110/bin/../lib/gcc/avr/4.3.3/../../../../avr/bin/ld.exe: section.version [00007ffe - > 00007fff] overlaps section .text [00007000 -> 0000817b]"
So i changed in the makefile "--section-start=.text=0x7000" to "--section-start=.text=0x817b" and then it compiled properly but my .hex file size is around 4800 bytes instead of 4000 with the old optiboot. 
So do you think it's possible to remove some files from optiboot to have more space ?

matthijskooijman, 2017-06-01:
> "C:/winavr-20100110/bin/../lib/gcc/avr/4.3.3/../../../../avr/bin/ld.exe: section.version [00007ffe - > 00007fff] overlaps section .text [00007000 -> 0000817b]"

This means the code (.text) is too big for the bootloader area (which starts at 0x7000 and runs up to 0x7fff, with the last two bytes reserved for the version number). To fix, you should enlarge the bootloader area (if possible, there's only a few sizes available) by changing the fuse settings and *lowering* the .text start address accordingly.

Zanatoos, 2017-06-01:
I see. But i was already on the maximum size for the bootloader, so i went back to the old optiboot. And after lowering LED_START_FLASH=20 to 1 or 0 it works so i feel quite stupid now. 

Thanks for your help anyway !