arduino/Arduino #6345: missing documentation: list of the Arduino board types + print the current one

tofrnr, 2017-06-04:
hello,
where can I find the list of the available Arduino board or MCU types (e.g., for Uno, Nano, Mega, M0, Due, but also supported boards like ESP8266, and how con I print the current board / MCU type from a running program to check the settings?
I need that overview and the current type for writing cross-board-compatible code for pin numbers and interrupt numbers, e.g. by
#ifdef  
#ifndef 
#elseif

facchinm, 2017-06-05:
Hi @tofrnr ,
unfortunately there is no such list... If you want to target specific boards you can use the ```
#ifdef ARDUINO_$PROVIDER_$BOARDNAME``` or the architecture generic ```#ifdef ARDUINO_ARCH_$ARCHTECTURE``` but if you want to target a family of microcontrollers (like atmega328p) you'll need to take a look at the compiler's specific defines. 
The fastest way to create this list is compiling with verbose output for all the boards you want to target and take a look al the generated commandline.

closed 2017-06-05 by facchinm

tofrnr, 2017-06-05:
why do you close that topic? it's not resoved !
please reopen!

tofrnr, 2017-06-05:
which constants for board and MCUs are used by Arduino?

cmaglie, 2017-06-05:
This is a question for the forum: https://forum.arduino.cc/

Please also search for already answered questions, I'm pretty sure that someone else already discussed the same question.

tofrnr, 2017-06-05:
no, this is a request for adding the (missing) documentation!

tofrnr, 2017-06-06:
@facchinm, 
please reopen this topic!
It is not resolved!
tbh, I consider your attitude to be quite rude and unrespectful: I reported this missing documentation about board and mcu type constants to be provided subsequently, and it's not been done so far.
If that ionce will have been published, I'll immediately agree to close this topic, don't worry, but so far nothing like that is visible and thus this topic has to be remained opened!.

facchinm, 2017-06-06:
@tofrnr ,
first of all, there is no explicit request to add documentation in the first post; you only said you need that for your library. Then, maintaining such list is a PITA and doesn't add any value to the Arduino ecosystem, which is always trying to be cross-platform by hiding such details at core level. If you really need to access low level functionalities (for example, to perform high speed operation on pins) you can take a look at @PaulStoffregen 's OneWire library (https://github.com/PaulStoffregen/OneWire/blob/master/OneWire.h#L61) which contains most of the needed defines. If you need to target a specific board, I repeat, the define is `#ifdef ARDUINO_$PROVIDER_$BOARDNAME` and I believe no-one wants to maintain a list of hundreds boards which all refer to this simple pattern.

bengtmartensson, 2017-06-06:
It appears as that what the O.P. wants can be achieved with extraction of information from the boards.txt files, possibly together with some formatting. Try 

    grep -h build.board. /usr/local/arduino/hardware/arduino/avr/boards.txt ~/.arduino15/packages/*/hardware/*/*/boards.txt | sed -e s"/[^=]*=//"|sort -u

as a first start (on typical Linux installation).(Can use cygwin if on Windows.)

tofrnr, 2017-06-06:
@facchinm,
I doubt vehemently that "it doesn't add any value to the Arduino ecosystem", and I don't care if you think it perhaps wouldn't. It is a documentation request and it's impertinent to close this request as long it is not resolved and provided, no matter what you think!

PaulStoffregen, 2017-06-06:
Actually, what Martino thinks does matter.  He's one of the very few active Arduino developers.  You aren't.

tofrnr, 2017-06-07:
so then please tell how to poll from a general program for AVR and ARM and Nodemcu boards
- the current number of maximum available digital pins
- the current number of maximum available analog pins
- which D and A pin numbers in between are reserved for  I2C, SPI, UART, i.e.
- the pin numbers for I2C +  SPI  
- which pins are available for pinchange interrupts, timer interrupts, and for pwm

for a simple example:
can I analogRead A4 (Mega, Due, M0) or is it reserved for I2C (Nano) or not at all (nodemcu)?
is there a  pin 15 (Mega, Due, nodemcu) and is it free for digital r/w (nodemcu)?
is A2 available (not on nodemcu)
is A6 available (Nano, Mega, Due, not Uno+nodemcu)
is pin 52 available (Mega, Due) and can I r/w to pin 51 (Due) or not (SPI on Mega)?

if there was a list of board and MCU type constants like requested then one could determine that easily on his own.

And reopen this topic because it's not resolved!

per1234, 2017-06-07:
@tofrnr you're being extremely rude. This is a place to report bugs with the Arduino IDE. As was already suggested, if you want help you need to post your questions on the forum. Hundreds of people get an email every time you make a pointless comment here. Please stop wasting the time of the Arduino developers so they can instead work on the things that benefit the entire Arduino community.

The board macros are already documented here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#boardstxt
There's no point in trying to maintain a full list. If you take the time to understand how the `build.board` property set in boards.txt is used you can easily determine the macro for any board.

As for the MCU specific macros, that's outside the scope of Arduino. A list of the AVR macros is found here:
http://www.atmel.com/webdoc/AVRLibcReferenceManual/using_tools_1using_avr_gcc_mach_opt.html
for MCU specific macros of other architectures you will have to do some research of your own.

tofrnr, 2017-06-07:
no, I was asking nicely to publish the list but my request was immediately closed in an extremely rude manner without solving the issue.
That was the only rude thing over here.

anyway, so is it true that its possible to determine during runtime if the current board is a Nano by
#ifdef __AVR_ATmega328__
and the Uno by
#ifdef __AVR_ATmega328P__
?
and if so, which are the constants for M0, Due, and nodemcu ?