arduino/Arduino #6350: Version 1.8.2 three times slower to launch

technoblogy, 2017-06-05:
I've been comparing the time to launch 1.6.8, 1.8.1, and 1.8.2 on a MacBook Pro (Mac OS X 10.11.6).

Both 1.6.8 and 1.8.1 take 4 seconds, but 1.8.2 takes 13 seconds, over three times slower.

Any suggestions why this might be?

PaulStoffregen, 2017-06-05:
Maybe the newer JRE?

gilbertoverg, 2017-06-07:
I have the same issue, for me the launch time for 1.8.2 and 1.8.3 is 27 seconds

bperrybap, 2017-06-08:
Some of the problem (maybe most of it) is due to something in the IDE that ends up scanning the users sketchbook libraries and cores multiple times. It does this on startup or any time you switch cores.
I don't understand multiple scans would ever be necessary.

This has existed for many years but seems to have gotten worse recently.
It gets particularly slow when your sketchbook directory is across a network mount.

In the past it was very obvious for me as certain IDEs had some foolish naming restrictions on library directory names and I would have to answer the same prompt for a given library directory name up to 9 times for each directory it didn't like on startup or whenever I switched cores.
i.e. I'd get a slew warnings for directory names it didn't like - each of which must be individually acknowledged, and then this same sequence would repeat over and over again until the IDE finally started. 
It was a total pain.
The new IDEs no longer have the restrictions on directory names and hence don't toss up the warnings that you must acknowledge but they seem to still do the multiple scans of the users sketchbook libraries.

gilbertoverg, 2017-06-09:
@bperrybap I agree with you, I can see Arduino scanning libraries during startup with:
```sudo fs_usage pid <Arduino pid>```

technoblogy, 2017-06-09:
Thank you for confirming that I'm not the only one who has found this. I definitely think it would be worth the developers spending a bit of time to fix this.

cmaglie, 2017-06-14:
The multiple scanning of the libraries can be surely optimized, but it has been like this well before 1.8.1, IMHO the cause of the slow down should be searched elsewhere.

bengtmartensson, 2017-06-14:
I have found that "phony" serial ports (e.g. Bluetooth virtual such) can be startup time culprit. Can this be an issue here?

f3rland, 2017-06-20:
Thanks for pointing that out @bengtmartensson 
I had multiple "VirtualBox Host-Only Ethernet Adapter #x" created by Virtual box/Vagrant and Arduino IDE (1.8.2) took nearly 1 minute to load before I removed all of those adapter. Now 3 seconds top.

gilbertoverg, 2017-06-28:
I've tried the test build from #6419 but it still takes 35 seconds to launch

apphead, 2017-09-05:
Issue still present in 1.8.4

jaygarcia, 2017-11-03:
Want to confirm that this is still present in 1.8.5. =)

facchinm, 2017-11-03:
@jaygarcia could you test 1.9.0 beta? It should contain an additional fix (5c61508) that should solve the issue. Thansk!

gilbertoverg, 2017-11-03:
1.9.0 beta solves the issue for me

technoblogy, 2017-11-03:
I confirm that 1.9.0 beta launches in 4 seconds, matching the behaviour of 1.6.8 and 1.8.1. That's great news - a big thank you to everyone who worked to achieve this!

SuperTango, 2017-12-06:
@jaygarcia dunno if this still matters, but I see this problem in 1.8.5.

jaygarcia, 2017-12-06:
@SuperTango this seems to be fixed in 1.9 beta: https://www.arduino.cc/en/Main/Software#beta

This ticket can most likely be closed once 1.9 is released to prod.

Fantastic work everyone and thank you so much for fixing this! <3

jaygarcia, 2017-12-06:
/cc @facchinm

commit fb2b0980 (2017-10-18): Wrap the discoverers into separate threads

Fixes #6350 (the regression was introduced with b2241dadf8f7777564bc5595265caa53a1fec399 )
Thanks git-bisect :)
commit 51fbe97d (2018-01-23): Wrap the discoverers into separate threads

Fixes #6350 (the regression was introduced with b2241dadf8f7777564bc5595265caa53a1fec399 )
Thanks git-bisect :)
commit 52c829fb (2018-05-02): Wrap the discoverers into separate threads

Fixes #6350 (the regression was introduced with b2241dadf8f7777564bc5595265caa53a1fec399 )
Thanks git-bisect :)