arduino/Arduino #6352: Cann`t come out from [while ();]

miladKhorasani, 2017-06-06:
In these program when i come in while than it is impossible to come out, i make the while condition false in interrupt.... 
let me explain you the more interesting issue that i convert the same code to codevision environment after compile and upload everything gone right ..... 

//In The name OF God 
//Project : Phantom Haptic device 
//Designed By MiladKhorasani 
//*********************************************History***************************************
//  01: 96/3/14  V.1 By Khorasani Make First Version 
//  02: 96/3/15  V.1 By Khorasani Make First Version - make diffirent Time(500ms-5sec-10sec) 
//******************************************Pin Assignment************************************
#define ledPin 13
#define ledPin1 8
#define ledPin2 6
//*****************************************Constants*******************************************

     

//Bit Setter & Resetter Definitions
#define bit0        0b00000001
#define bit1        0b00000010
#define bit2        0b00000100
#define bit3        0b00001000
#define bit4        0b00010000
#define bit5        0b00100000
#define bit6        0b01000000
#define bit7        0b10000000
//***************************************BitField Definitions******************************************
//Task Flags
//rBitControl1
#define   bIdleLoop         bit0
#define   bCountTask     bit1
#define   bLED                  bit2
#define   bLed5sec          bit3
#define   bLed10sec       bit4
//****************************************Variables***************************************
int var;
int timer1_counter;
unsigned char rBitControl1;
unsigned char rBaseTimeCounter ;
unsigned char rCounter5sec ;
unsigned char rCounter10sec ;
//******************************************Functions Declerations*************************************

//******************************************Main Body of Program*************************************

void setup()
{
  pinMode(ledPin, OUTPUT);
  pinMode(ledPin1, OUTPUT);
  pinMode(ledPin2, OUTPUT);

  // initialize timer1 
  noInterrupts();           // disable all interrupts
  TCCR1A = 0;
  TCCR1B = 0;

  // Set timer1_counter to the correct value for our interrupt interval
  //timer1_counter = 64911;   // preload timer 65536-16MHz/256/100Hz
  //timer1_counter = 64286;   // preload timer 65536-16MHz/256/50Hz
  timer1_counter = 34286;   // preload timer 65536-16MHz/256/2Hz
  
  TCNT1 = timer1_counter;   // preload timer
  TCCR1B |= (1 << CS12);    // 256 prescaler 
  TIMSK1 |= (1 << TOIE1);   // enable timer overflow interrupt
  interrupts();             // enable all interrupts
}

ISR(TIMER1_OVF_vect)        // interrupt service routine 
{
     rBitControl1 = rBitControl1 &(~ bIdleLoop) ;
     rBitControl1 = rBitControl1 | bCountTask ;
     TCNT1 = timer1_counter;   // preload timer
  
}

void loop()
{
 if( rBitControl1 &  bCountTask  )
    {
      runCountTask();
      rBitControl1=rBitControl1 &(~ bCountTask);
    }

 if( rBitControl1 &  bLED   )
    {
      led();
      rBitControl1=rBitControl1 &(~ bLED); 
    }
    
 if( rBitControl1 &  bLed5sec   )
    {
      led5sec();
      rBitControl1=rBitControl1 &( ~ bLed5sec ); 
    }
 if( rBitControl1 &  bLed10sec   )
    {
      led10sec();
      rBitControl1=rBitControl1 &( ~ bLed10sec ); 
    }

  rBitControl1= rBitControl1 | bIdleLoop ;   
  while (rBitControl1 & bIdleLoop)
    {
      __asm__("nop\n\t");
	  __asm__("nop\n\t");
	  __asm__("nop\n\t");
	  __asm__("nop\n\t");
    }

}

void runCountTask (void)
{
  rBaseTimeCounter++;
  if (rBaseTimeCounter >= 2) 
  {
    rBaseTimeCounter = 0;
    rBitControl1= rBitControl1 | bLED ;
    rCounter5sec++;  
    if (rCounter5sec >= 5)
     {
       rCounter5sec = 0 ;
       rBitControl1 = rBitControl1 | bLed5sec ;
     
         rCounter10sec++;
         if(rCounter10sec>=2)
          {
         rCounter10sec = 0 ;
         rBitControl1 = rBitControl1 | bLed10sec ;
          }
      }
   }
    
}

void led()
{
  digitalWrite(ledPin, digitalRead(ledPin) ^ 1);
 } 
void led5sec()
{
  digitalWrite(ledPin1, digitalRead(ledPin1) ^ 1);
} 

void led10sec()
{
  digitalWrite(ledPin2, digitalRead(ledPin2) ^ 1);
}

cmaglie, 2017-06-06:
This is not the place to ask for support on your own project, please use the forums:

https://forum.arduino.cc/

closed 2017-06-06 by cmaglie

miladKhorasani, 2017-06-06:
Hello , i didn't ask support, i xplain the issue of arduino that i faced ...

cmaglie, 2017-06-06:
> I didn't ask support

so what's the purpose of your message then?

miladKhorasani, 2017-06-06:
Explain the issue of arduino IDE .

cmaglie, 2017-06-06:
Why are you so sure that it's a problem with the Arduino IDE?
Could it be an error in your sketch? Have you discussed the sketch with some experts on the forum?

miladKhorasani, 2017-06-06:
Because i test the same sketch in codevision environment and it was okey ,
but I haven't discussed on the forum.

cmaglie, 2017-06-06:
> Because i test the same sketch in codevision environment and it was okey.

This doesn't mean anything, there is surely an error in your sketch, at the minimum you don't use the keyword `volatile` that is needed when a variable is shared between the main loop and an interrupt subroutine.

Please consult the forum before posting here any question about a sketch.

miladKhorasani, 2017-06-06:
volatile Made everything okey.
Thank you and sorry for made that issue.