arduino/Arduino #6355: arduino download new packages has error

bigelite, 2017-06-07:
Hi. I have installed a fresh copy of the IDE but continue to get this error when opening the boards manager. I have followed instructions on the forum to rectify the problem but nothing works.
I am running Windows 10.
I can access this URL from my browser.
I am able to install some of the boards that are listed in the boards manager, but most give errors when trying to install.

[image]

Error downloading http://downloads.arduino.cc/packages/package_index.json

I've deleted  staging/packages folder.
and i tried  this one :
Copy: setx _JAVA_OPTIONS -Djava.net.preferIPv4Stack=true

i anyone can help me?

facchinm, 2017-06-07:
Hi @bigelite ,
if you click on the link (http://downloads.arduino.cc/packages/package_index.json) are you able to see the content? It appears to be a network problem of some sort (or an incomplete proxy configuration)

bigelite, 2017-06-12:
yes i can see the JSON into that link.
[image]
here is windows proxy setting,

and 
[image]
this is arduino preferences network.
[image]

but error is still exist!

Squirrelwolf, 2017-06-14:
I suppose this is related, considering previous similar problems. I'm trying to install Intel Curie Board files, but I get "Error downloading https://downloadmirror.intel.com/25470/eng/arc-toolchain-osx-arcem-1.0.1.tar.bz2"

feikname, 2017-06-16:
Perhaps the the problem is caused by a firewall or an antivirus?

Squirrelwolf, 2017-06-16:
Regarding my problems: The server seems to be working again. (And no firewall or antivirus here.)

bigelite, 2017-06-17:
does it occur using by kaspersky antivirus? how can i prevent antivirus? @feikname

facchinm, 2017-06-19:
@bigelite your AV should have some sort of "[whitelist](https://support.kaspersky.com/12677#block2)", try adding the IDE to this list and please report if it worked, thanks!

per1234, 2017-08-22:
@bigelite did you ever try whitelisting the Arduino IDE in your antivirus? If so, did it solve this issue for you?

per1234, 2018-01-16:
Closing due to lack of response from @bigelite.

closed 2018-01-16 by per1234