arduino/Arduino #6358: Request: adding new libraries (AJSP/C++Tasks/LEDMatrixDriver)

bartoszbielawski, 2017-06-07:
Hi,

I'd like to request to have the following libraries added to Library Manager:

- https://github.com/bartoszbielawski/AJSP - streaming JSON parser
- https://github.com/bartoszbielawski/C-Tasks - a C++ library for FSMs/"coroutines"
- https://github.com/bartoszbielawski/LEDMatrixDriver - Library for MAX7219 based matrix displays

Regards,
Bartosz Bielawski

cmaglie, 2017-06-12:
I've added LEDMatrixDriver and AJSP.

About C-Tasks the library is named `C++Tasks` but the `+` character is not allowed to be in the library name, may you rename the library to remove the `+`?

PaulStoffregen, 2017-06-12:
Your examples aren't in folders and structured as Arduino sketches (at least one .ino file with the same name), so they won't show up in Arduino's File > Examples menu.  Maybe you care, maybe not?

bartoszbielawski, 2017-06-12:
@cmaglie  I will rename it (+ -> P) and I will let you know, thank you.
@PaulStoffregen  Yes, I'm aware I didn't add any example sketches yet. I'm planning to add them in one of the future releases.

bartoszbielawski, 2017-06-12:
@cmaglie I have renamed the C++Tasks to CPPTasks, here's the new link: https://github.com/bartoszbielawski/CPPTasks

cmaglie, 2017-06-14:
Done!

closed 2017-06-14 by cmaglie