arduino/Arduino #6359: IDE window: CTRL-MouseWheel for zoom

snobu, 2017-06-07:
The inability to zoom in/out the IDE window is slightly annoying, i'm sure i'm not the only out there bothered by it. What's the development effort required to implement either `CTRL-MouseWheel` and/or `CTRL-+`, `CTRL-minus`? Do we have to move away from Java? :) Or just bind to some events?

bengtmartensson, 2017-06-08:
Possibly Java 9, due next month, will bring something here?

facchinm, 2017-06-09:
That's a good suggestion , thanks for posting! It should be fairly easy to trap the event inside the Editor view and apply the preference. Hot-reloading could be a bit more difficult but feasible indeed. If anyone wants to try coding a solution we'll be super happy to test it :smile:

skishida, 2017-08-23:
Now, font size may become a negative value in #6551 change and at that time, the IDE will not start.
The temporary fixing approach is edit `editor.font` in `preferences.txt`.

PR #6551 (2017-08-01): Add font size shortcuts (menu, keyboard and mouse)
Incorporates and expands https://github.com/arduino/Arduino/pull/6394

Fixes  #6359
commit 5427f94b (2017-08-28): Do not allow negative font resize

Fix #6359 (again)
closed 2017-08-28 by cmaglie