arduino/Arduino #6365: IDE 1.8.3 has old java.exe that Kaspersky flags as a vulnerability...

coderpop, 2017-06-09:
Hi,

Seems that the version of java.exe supplied with 1.8.3 is outdated and flagged by Kaspersky as a vulnerability.

Can a newer version be included? The one currently supplied is from last year whereas the latest is from April this year.

Thanks.

NicoHood, 2017-06-09:
That might also apply to the linux version. Its the reason we never use static code linkage whenever possible. Please do not ship java with arduino, if its not maintained. Also especially because a lot of users use older versions of the IDE and don't upgrade. You have to protect them. Let them install java themselves.

facchinm, 2017-06-12:
Hi Nico,
java is being bundled on releases for every OS since 1.6.5 to overcome some (really difficult to reproduce) bugs due to really outdated or "alternative" jre installations. We try to keep the package updated so thanks for reporting the availability of a new versions, I'll update them today so next nightly build will contain them.

closed 2017-06-12 by facchinm