arduino/Arduino #6369: Request: Native Atmega328-AU Support

ArcAIN6, 2017-06-11:
While the atmega328p is popular, the non-picopower version is just as useful. The humble Atmega328 requires the user to manually edit the hardware config file. 

The problem with this, is people attempting to use these MCU's for the first time, run into an issue they rarely understand, or have difficulty tracking down, that is the IDE throwing an error about the ID. 

It would be nice if arduino IDE simply just worked with these chips, without putting the user through the arduous task of trying to track down why the ID is different. 

I myself ran into this situation, and spent almost 2 weeks googling before the answer presented itself. Most notably, most of the search returns simply dealt with the atmega328p being wired wrong as the culprit, and this simply isn't the case. People are so used to the 328p, that they simply overlook the p is missing, and assume the person with the issue is talking about that MCU. 

Thank you for taking the time to read this, and i hope this is helpful to someone searching for an answer.

per1234, 2017-07-06:
I don't see a non-hacky way to make it "just work". You have to know what part you're using and use a hardware package that supports your hardware. There's no reason for Arduino AVR Boards to do this because none of the Arduino AVR Boards use ATmega328. For ATmega328 I recommend the excellent MiniCore: https://github.com/MCUdude/MiniCore.

Arduino definitely doesn't help this problem by frequently calling the ATmega328P "ATmega328". You can see this in multiple places in on the website and even in the Tools > Processor menu for Diecimila, Nano, Mini, BT, LilyPad, and Pro Mini!!! So this certainly could be partially avoided by fixing the documentation.

ArcAIN6, 2017-07-08:
Thank you for the reply, and the point to minicore, i hadn't seen that before :)

closed 2018-06-04 by per1234