arduino/Arduino #6370: feature request: enable to load .h and .c files (from arbitrary  libs)  into Arduino IDE

tofrnr, 2017-06-11:
feature request: 
enable to load .h and .c files (from arbitrary /proprietary libs ) into the Arduino IDE
When designing, creating, and reworking libs for a current project it's extremely cumbersome to use external editors and the Arduino edtor side by side.
Instead, it shoud be possible to open the .h and .c files in extra tabs additionally to the tab for the  current  .ino main file.

per1234, 2017-06-11:
Duplicate of https://github.com/arduino/Arduino/issues/3512

From https://github.com/arduino/Arduino/blob/master/CONTRIBUTING.md#1-before-using-the-issue-tracker:
>But check the following boxes before posting an issue:
...
>- [ ] `Your issue is not a duplicate.` So search for similar open and closed issues and pull-requests.

closed 2017-06-12 by facchinm

tofrnr, 2017-06-12:
I don't care if it's a duplicate or not, it's finally not resolved.

tofrnr, 2017-06-13:
as stated elesewhere:
editing entire libraries is not my intention but opening arbitrary .c and .h files (even if part of a Arduino "library", or even if not), and so I would search for "open .c and .h files", not "editing libraries". Apart from that "editing libraries" is misleading as libraries actually are .so and .a files in C. So I never would come to the idea to search for "edit libraries" to see if one ever wanted to load a .c or a .h file into an extra IDE tab.

To clarify:
I want to open such standard C files from where ever they may come (e.g., mostly from other library code but optionally also from other C, C++, or ino sources ) and want to #include them into my code  and store the files at another place. I do not want to create entire "libraries", and I do not want to edit entire existing libraries, neither in the sense of standard C (.so, .a) nor of the way how Arduino constructs them out of other files in proprietary subdirectories plus extra properties and keywords files.

In that respect my intention is perhaps a small subset of "editing libaries" which is actually out of my scope, **and so I did not search for "edit libraries"** before posting this and OTOH **"load .h and .c files  into Arduino IDE" does not match to any different existing and still opened and still unresolved topic.**