arduino/Arduino #6371: how to setup BLE multiple services and multiple characteristics UUID?

guoxuxu, 2017-06-12:
I'm now have the problem of setting up BLE with 6 services and each service has 10 characteristics. I resigned the service UUID as human-interaction 0x1124, and 0x1125,1126,1127,1128,1129. and each characteristics as 0x0001,0002,0003,0004,0005 and so on. The problem is that each of them work when used with only one characteristic, but out of use when put the ten together into use. 

I can't find any useful document for reference, the examples on the arduino website are all the same with only on service and only one characteristic without any reason why the UUID was setup like this. I could find some document introduce what UUID to use for certain service but no specification for how to set up several same services and resign UUID for them.

please tell me how to resign UUID for multiple services and multiple characteristics, for example 6 services and each service 10 characteristics.

thank you

guoxuxu, 2017-06-12:
ps: I could use 0x 0001-0x0008 for each service but couldn't add more. Is this the limit number of characteristics for each service?

facchinm, 2017-06-12:
Hi @guoxuxu ,
your report is missing both the board and the libraries you are using. We can't really help (or give you the right directions where to get help) if we don't have these information. Thanks.

guoxuxu, 2017-06-13:
Thank you for your reminder, I used The Arduino 101 board that is developed by Intel and the library I used was CurieBLE.h.

CurieBLE of Arduino: https://www.arduino.cc/en/Reference/CurieBLE

guoxuxu, 2017-06-13:
I followed the example on https://www.arduino.cc/en/Reference/BLECharacteristicConstructor
But actually in the Arduino software, the same examples given by the software was different from the website, it used "addCharacteristic" instead of using "addAttribute" .For me, "addCharacteristic" is more clear because I can distinguish adding Characteristic from adding Service by applying "addCharacteristic" and "addService", while it's vague to use "addAttribute" for both adding Characteristic and adding Service.

guoxuxu, 2017-06-13:
This is the hardware I used: Intel Curie Module Based Board https://www.element14.com/community/docs/DOC-82913
This is the development board I choosed in the Arduino software: Arduino/Genuino 101.

facchinm, 2017-06-13:
Hi @guoxuxu ,
thanks for the info! The problem may be related with the limited amount of RAM available to on the Curie core for the Arduino processor. There is a lot of work in progress on fixing this (like [this](https://github.com/01org/corelibs-arduino101/commit/562cd84678190f978c4e6921a587d4a269172371) commit) but in the meantime a safe maximum is 6 to 8 characteristics. If you really can't slim down your sketch, please open an issue [here](https://github.com/01org/corelibs-arduino101/issues) or leave a message on the [forum](http://forum.arduino.cc/index.php?board=103.0) to get some further assistance. Thanks!

closed 2017-06-13 by facchinm

guoxuxu, 2017-06-13:
Thank you for your answer, I actually tried out the maximum 8 characteristics exactly like you said. Your answer was the same with my primary guess, but I would like to ask that if the BLEService and BLECharacteristic functions or classes of Curie library could be slimed down in the future? For example, delivering the number of characteristics or services as a parameter to the function or class, instead of declaring the function or class again and again. Because all the arduino hardware actually have limited RAM and I couldn't cut down my demands on the service.

guoxuxu, 2017-06-13:
For example, I want the declaring "BLEShortCharacteristic batteryLevelChar("0001",BLERead | BLENotify);" in an array form as "String UUIDArray[] = {UUID1, UUID2, UUID3, UUID4};BLEShortCharacteristicArray batteryLevelChar[] (UUIDArray,BLERead | BLENotify);". I don't know the fundamental code of BLEShortCharacteristic, so I venture to guess if this Array form could reduce the RAM by iterating the same function with different parameters.

AmishaKuwarbi, 2020-02-22:
Hey, I used ESP32 with Arduino IDE in which I created two characteristics of unique UUID sharing the same service UUID. The code should include individual setting of value and scan request + advertising for each characteristics. After establishing the server, i used a BLE scanner app to scan my ESP and I was able to observe two characteristics in a single server
[image]