arduino/Arduino #6372: feature request, not resolved: enable to load .h and .c files (from arbitrary libs) into Arduino IDE

tofrnr, 2017-06-12:
feature request:
enable to load .h and .c files (from arbitrary /proprietary libs ) into the Arduino IDE
When designing, creating, and reworking libs for a current project it's extremely cumbersome to use external editors and the Arduino edtor side by side.
Instead, it shoud be possible to open the .h and .c files in extra tabs additionally to the tab for the current .ino main file.

per1234, 2017-06-12:
Second duplicate posting of this issue:
https://github.com/arduino/Arduino/issues/6370

Extremely rude in this issue:
https://github.com/arduino/Arduino/issues/6345

Time to ban this user.

tofrnr, 2017-06-12:
yes, perhaps, maybe, maybe not, but it's not resolved yet and it still remains a feature request.

matthijskooijman, 2017-06-12:
@tofrnr, As pointed out by @per1234, #3512 requests this exact same thing. There is no point in having two open issues for the same request, which is why your previous request was closed (and why I'll also be closing this one). The other request is indeed old, since this is apparently not a priority for the developers and also a non-trivial change that needs some proper thought and planning.

If you have additional suggestions or information, feel free to leave a comment with that issue (but if it's only "I would also like this", then rather use a +1/thumbsup reaction on the first post, not a new comment).

closed 2017-06-12 by matthijskooijman

tofrnr, 2017-06-12:
@matthijskooijman,
thank you for your reply. Now I see your point, and above all, your reply now sounds much more reasonable and much more polite and respectful than the rude and  ignorant attitude of facchinm when  just closing my topic without further comment and without recognizable justification, and even more than the insulting and accusing post of per1234which is so extremely bad-mannered, that this member should be permanently banned. 
So if the former request is still  not out of your scope, I hope that it will be then resolved in a near future.

matthijskooijman, 2017-06-12:
@tofrnr, with the dozens of new issues being filed each week, there is limited time for elaborate replies and often duplicate issues are simply closed with a reference to the duplicate. There is certainly a responsibilty for anyone reporting issues to get familiar with the common practices of bug reporting, and look for open issues themselves (which doesn't always mean you'll find them, of course). Since recently, these kind of things are written down in CONTRIBUTING file which is linked above the new issue form and will be expanded to contain more helpful information.

As for perceived rudeness, simply re-opening another issue instead of asking for clarification is often seen as rude as well, but most of the rudeness here stems from coming from a different angle and is mostly misunderstanding. I'm pretty sure that nobody intended to be rude here :-)

In any case, good to have things cleared up now. I also hope we'll resolve the library editing issue in the future.

oldmanegan, 2017-06-12:
From someone that learned rapidly that "stark" brusqueness (from James in my instance) must not be taken personally and that there are many cultures represented here, I recommend that Matthijs Kooijman's explanation be posted as appropriate to the entry point to the issues list so that anyone making a submission is somewhat pre-knowledgeable of potential reactions (as a form of submitter's psyche flack vest) while also cautioned to not judge someone based on their possibly cryptic and negative response.

tofrnr, 2017-06-12:
entering into the search field 
load .h and .c files into Arduino IDE 
there s no single match to any related  topic!

per1234, 2017-06-12:
@oldmanegan the closest thing to the "entry point" would be [CONTRIBUTING.md](https://github.com/arduino/Arduino/blob/master/CONTRIBUTING.md), which GitHub displays a link to whenever someone begins the process of creating an issue or pull request. I currently have an open pull request for some improvements to the current CONTRIBUTING.md and would be interested in any specific suggestions you might have for the content but we should confine further discussion on this topic to the comment thread of that pull request: https://github.com/arduino/Arduino/pull/6357

@tofrnr the obvious search terms are ["edit libraries"](https://github.com/arduino/Arduino/issues?page=3&q=edit+libraries). You may notice that you have to look through three pages of search results before finding that issue. This is why duplicate, invalid, or vague issue reports are such a huge problem. I submit quite a few issue reports here and I always try to do my best to avoid submitting duplicates. With over 5000 issue reports and over 1000 pull requests to search through this can sometimes be a lot of work and will become more so over time. For this reason, I have a long list of legitimate issues on my "to do" list that I haven't found the time to report yet. Of course valid entries are welcome but a large number of that 5000 are invalid. You've added 3 or 4 to that number already.

Regarding the subject of this issue report, if you check the original issue you'll see I fully agree that this would be a useful feature and in fact I already use the Arduino IDE for editing my libraries.

tofrnr, 2017-06-12:
editing entire libraries is not my intention but opening arbitrary  .c and .h files  and so I would search for "open .c and .h files", not "editing libraries". Apart from that "editing libraries" is misleading as libraries actually are .so and .a files in C. So I never would come to the idea to search for "edit libraries" to see if one ever wanted to load a .c or a .h file into an extra IDE tab.

per1234, 2017-06-12:
>libraries actually are .so and .a files in C

@tofrnr so now you are going to pretend ignorance just to avoid admitting you're wrong? If you have a .c, .h, etc. file in the root of the sketch folder *the Arduino IDE already opens it*. If those files are elsewhere _that's considered a library in the Arduino world_. I refuse to believe you don't already know this. You even use the word "libs" in your issue:
>enable to load .h and .c files (from arbitrary /proprietary libs )

PaulStoffregen, 2017-06-13:
This combative tone really isn't helping resolve what should be just a simple miscommunication.

I'm pretty sure what you want is the ability to view or edit the source code within the libraries that are getting compiled into your sketch.  How editing would work for libs builtin libs like Servo or architecture specific libs like Wire is a good question.

But such a hostile tone hardly inspires much interest in finding the answers to those sorts of issues.  You're really doing your cause more harm than good by speaking this way.

matthijskooijman, 2017-06-13:
@tofrnr, editing arbitrary .c and .h files does seem like a (slightly) different (though partly overlapping) thing than editing libraries, but it would probably make sense to implement them at the same time, so also keep track of both requests in the same issue. Perhaps you could add a comment along these lines to the other issue?

In the meanwhile, these comments aren't really going anywhere, so I'm gonna lock this issue to prevent further discussion.