arduino/Arduino #6374: A new library

Onno-Dirkzwager, 2017-06-12:
Could you please add this library to the "Manager"

https://github.com/iotappstory/ESP8266-Library-V1.0

Any feedback or recommendations are welcome!

per1234, 2017-06-12:
You need to create a tag or release with a version matching the value of the `version` property in library.properties:
https://github.com/iotappstory/ESP8266-Library-V1.0/blob/master/library.properties#L2

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Onno-Dirkzwager, 2017-06-12:
Thankyou for your feedback.
We just did: https://github.com/iotappstory/ESP8266-Library-V1.0/releases

Onno-Dirkzwager, 2017-06-13:
Per1234, 
thanks for the feedback. We made the tag / release: 
https://github.com/iotappstory/ESP8266-Library/releases/tag/1.0.0

And removed -V1.0 from the lib name:
https://github.com/iotappstory/ESP8266-Library

cmaglie, 2017-06-14:
Done!

closed 2017-06-14 by cmaglie

Onno-Dirkzwager, 2017-06-14:
Thank you sir!