arduino/Arduino #6378: IDE enhancement request: better search function: only find, find+replace, find only whole words

tofrnr, 2017-06-13:
IDE enhancement request: 
please provide a better search function: 
only find, 
find+replace, 
extra option: find only whole words  

(keyword search for 
find search replace whole words  
did not show any matches!)

quequen, 2017-06-15:
Buenas tardes:solicite bajar IDE Arduino l.8.3 para Win-hice la donacion solicitada y logro de ingresar ba la misma para bajar dicho archivo,(adjunto recibo-
Arduino, LLC recibo [# 1352-0594-

tofrnr, 2017-06-16:
what? please post in English!

tofrnr, 2017-06-27:
why has this not been tagged as " Component: IDE / Type: Improvement" meanwhile?

per1234, 2017-07-06:
Closing as duplicate of https://github.com/arduino/Arduino/issues/3727.

closed 2017-07-06 by per1234

tofrnr, 2017-07-07:
per1234: that's weird: a referenced post once opened at  25 Aug 2015, still unresolved, now you close my report/topic although it's still  unresolved  - why the heck do you close it then?? Instead of closing unresolved issue you'd better resolve them !!

cmaglie, 2017-07-07:
@tofrnr 
the issue has been closed as **duplicate** of #3727, it has no sense to have multiple issue about the same topic. Please note that #3727 is still open.

> Instead of spending time on searching so-called duplicates and closing unresolved issues you'd better use your time to resolve them

removing duplicates actually helps a lot to keep the issues tracker clean and to keep the all the informations into a single issue instead of having them scattered around many issues.

tofrnr, 2017-12-14:
is there meanwhile an update about the extra option "search whole words only" ?
without that option even the "replace all" button causes total mess!