arduino/Arduino #6381: HardwareSerial and general documentation issues

rkost, 2017-06-13:
So I just took a look into the core libraries and really wondered why there are no doxygen comments at all. That may be fine if there would be a documentation elsewhere, but as I searched for Stream::peek(), the official docs are saying "Returns the next byte (character) of incoming serial data without removing it from the internal serial buffer." the code is `virtual int read() = 0;` and in the HardwareSerial I was able to find a buffer that was defined like `unsigned char _rx_buffer[SERIAL_RX_BUFFER_SIZE];`.

So why is Stream returning int at all?

It would be that helpful if there where a strict rule for documentation inside the code. Best example: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Stream.h
There are comments one line before a definition, there are inline comments and even comments after the function definition.

In my eyes a nice documented code would look like it follows. And yes. Of cause it will add an enormous amount of lines but it will make live much easier. You can autocreate a documentation for the whole code and you do not need to dive into implementation to look what is being returned in special cases.

```c++
/*
  Stream.h - base class for character-based streams.
  Copyright (c) 2010 David A. Mellis.  All right reserved.

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA

  parsing functions based on TextFinder library by Michael Margolis
*/

#ifndef Stream_h
#define Stream_h

#include <inttypes.h>
#include "Print.h"

// compatability macros for testing
/*
#define   getInt()            parseInt()
#define   getInt(ignore)    parseInt(ignore)
#define   getFloat()          parseFloat()
#define   getFloat(ignore)  parseFloat(ignore)
#define   getString( pre_string, post_string, buffer, length)
readBytesBetween( pre_string, terminator, buffer, length)
*/

/**
 * @brief This enumeration provides the lookahead options for parseInt(), parseFloat()
 * The rules set out here are used until either the first valid character is found
 * or a time out occurs due to lack of input.
 */
enum LookaheadMode{
  /// All invalid characters are ignored.
  SKIP_ALL,
  /// Nothing is skipped, and the stream is not touched unless the first waiting character is valid.
  SKIP_NONE,
  /// Only tabs, spaces, line feeds & carriage returns are skipped.
  SKIP_WHITESPACE
};

/// a char not found in a valid ASCII numeric field
#define NO_IGNORE_CHAR  '\x01'

/**
 * @brief The Stream class.
 * For handling character streams.
 */
class Stream : public Print
{
[... 214 lines cut ...]

  /**
   * @brief Something useful here.
   */
  struct MultiTarget {
    /// string you're searching for
    const char *str;
    /// length of string you're searching for
    size_t len;      
    /// index used by the search routine.
    size_t index;     
  };
  
  /**
   * @brief This allows you to search for an arbitrary number of strings.
   * @param targets Something useful here
   * @param tCount Something useful here
   * @return index of the target that is found first or -1 if timeout occurs.
   */
  int findMulti(struct MultiTarget *targets, int tCount);
};

#undef NO_IGNORE_CHAR
#endif
```

rkost, 2017-06-13:
BTW: Of cause functions that have been inherited do not need to be documented a second time. One may write nothing in front of them or use `@copydoc` (https://www.stack.nl/~dimitri/doxygen/manual/commands.html#cmdcopydoc) or something.

Doxygen is that powerful...

oqibidipo, 2017-06-14:
>  * TODO: This should be const, shouldn't it? It should not make any changes to the stream.
  virtual int peek() = 0;

No, it shouldn't.
A counterexample from USB CDC Serial which does change the object:

[`avr/cores/arduino/CDC.cpp`](https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/CDC.cpp#L181) 
```
int Serial_::peek(void)
{
	if (peek_buffer < 0)
		peek_buffer = USB_Recv(CDC_RX);
	return peek_buffer;
}
```

matthijskooijman, 2017-06-14:
One can argue that it does not change the *visible* state of the object, just an internal detail.  The C++ language supports this notion with the `mutable` specifier (see http://en.cppreference.com/w/cpp/language/cv) which essentially allows marking members so they can be modified even on a const object (and thus from a const method).

Having said that, I'm not sure if that is sufficient justification to add const to `peek()`, since there might be subclasses that *do* want or need to change visible state in the `peek()` method.