arduino/Arduino #6385: [Library Manager] Question about adding a released library (Helium)

madninja, 2017-06-16:
I'd like to see  https://github.com/helium/helium-arduino added, but since the code relies on a submodule I have the travis builder create a package that includes the submodule as part of the release. 

See https://github.com/helium/helium-arduino/releases/tag/0.1.15 for an example. Helium-0.1.15.zip is the released zip with the git submodule included. 

It's not clear from the FAQ if the library manager automation script only picks up the github generated source zip (which excludes the submodule) or if there's a naming convention to get it to pick up a released bundle we built. 

Any hints appreciated!

cmaglie, 2017-06-22:
Out indexer picks the content of the git repositories from a clean `checkout` and zip it (more or less what github does with his generated zip-files). Your current setup, with git submodules, is not suitable for the library manager.

madninja, 2017-06-22:
@cmaglie understood. I'll fix up the repo to include the submodule sources directly. Should be ready shortly

madninja, 2017-06-23:
@cmaglie ok, I updated the repo to include the code of the submodule. Would you be able to get it into the library manager?

cmaglie, 2017-06-26:
Done!

closed 2017-06-26 by cmaglie