arduino/Arduino #6386: ESP8266 nodeMCU : while(1) at end of loop: runtime error, reset (on Uno ok)

tofrnr, 2017-06-16:
running the following code on an ESP8266 nodeMCU, the  while(1) at end of loop causes a runtime error, reset (on Uno ok though):

>Soft WDT reset
>
>ctx: cont 
>sp: 3ffef2a0 end: 3ffef4a0 offset: 01b0
>
> >>>stack>>>
> 3ffef450:  3ffe8438 00000001 3ffee448 402020a1  
> 3ffef460:  0000007f feefeffe 3ffee448 40202504  
> 3ffef470:  3fffdad0 00000000 3ffee448 3ffee46c  
> 3ffef480:  3fffdad0 00000000 3ffee464 402027b4  
> 3ffef490:  feefeffe feefeffe 3ffee480 40100114  
> <<<stack<<<
> 
>  ets Jan  8 2013,rst cause:2, boot mode:(1,7)

Uno: no error, ok;
instead delay(5000) for ESP8266: also no error, ok.

 

```
    // --------------------------------------
    // i2c_scanner
    //    
    // This sketch tests the standard 7-bit addresses
    // Devices with higher bit address might not be seen properly.
    //
     
    #include <Wire.h>

 
    void setup()
    {
      Wire.begin(D4,D5);  // ESP8266
      //Wire.begin();       // Uno
 
      Serial.begin(115200);
      while (!Serial);             // Leonardo: wait for serial monitor
      Serial.println("\nI2C Scanner");
    }
     
     
    void loop()
    {
      byte error, address;
      int nDevices;
     
      Serial.println("Scanning...");
     
      nDevices = 0;
      for(address = 0; address < 127; address++ )
      {

        if (address%16 == 0)  {   
          Serial.println();  
          Serial.print( (address+1)/16);
          Serial.print("  ");
        }

        
        if(address==0) {
           Serial.print("** ");
           continue;
        }
        
        Wire.beginTransmission(address);
        error = Wire.endTransmission();
     
        if (error == 0)
        {          
          if (address<16) Serial.print("0"); 
          Serial.print(address,HEX); Serial.print(" ");         
          nDevices++;
        }
        else if (error==4)
        {
          Serial.print("?? ");   
        }    
        else
        {      
           Serial.print("-- ");   
        }          
      }
      
      Serial.println();
      Serial.print("found: "); Serial.print(nDevices); Serial.print(" devices \n");
     
      //delay(5000);  /// <<< fine on all boards
      while(1);    // <<<<<<<<<< runtime error: reset on ESP8266 // Uno: ok
    }

```

per1234, 2017-06-16:
See https://github.com/esp8266/Arduino/blob/master/doc/faq/a02-my-esp-crashes.rst#watchdog
This is the intended behavior of the ESP8266 and does not represent a bug.

The ESP8266 core for Arduino is a 3rd party project. Any issues related to that project should not be reported here as it is unrelated to this repository. However, since this isn't actually a bug you should not report it to the esp8266/Arduino repository either.

sandeepmistry, 2017-06-16:
@tofrnr please see @per1234's comment above.

closed 2017-06-16 by sandeepmistry