arduino/Arduino #6387: Modify USB profile to remove serial port - just want keyboard (Leonardo)

unicorn79, 2017-06-16:
Hello! I'm using IDE 1.8.0 and Arduino leonardo. I try the standard USB HID keyboard sketch. Please tell me how to disable the serial port that appears when the device is connected (after the bootloader finish) - it's called Arduino Leonardo (COM 40) for example.  But i need, that there is only a USB HID keyboard and nothing else.
In this thread, dude seem to have decided this question:
[https://forum.arduino.cc/index.php?topic=262897.msg1856471#msg1856471](url)
but the structure of the USBDesc.h file in IDE 1.8.0 is different ...
Any ideas? Thanks!

facchinm, 2017-06-19:
Hi @unicorn79 ,
there is no easy way to get rid of the CDC interface because it's usually the only way to reprogram the board. Anyway, if you know what you are doing, a fast and hacky method is
```patch
diff --git a/hardware/arduino/avr/cores/arduino/USBCore.cpp b/hardware/arduino/avr/cores/arduino/USBCore.cpp
index e00fb028e..efbee7be5 100644
--- a/hardware/arduino/avr/cores/arduino/USBCore.cpp
+++ b/hardware/arduino/avr/cores/arduino/USBCore.cpp
@@ -376,8 +376,8 @@ bool ClassInterfaceRequest(USBSetup& setup)
 {
        u8 i = setup.wIndex;
 
-       if (CDC_ACM_INTERFACE == i)
-               return CDC_Setup(setup);
+       //if (CDC_ACM_INTERFACE == i)
+       //      return CDC_Setup(setup);
 
 #ifdef PLUGGABLE_USB_ENABLED
        return PluggableUSB().setup(setup);
@@ -467,7 +467,7 @@ static u8 SendInterfaces()
 {
        u8 interfaces = 0;
 
-       CDC_GetInterface(&interfaces);
+       //CDC_GetInterface(&interfaces);
 
 #ifdef PLUGGABLE_USB_ENABLED
        PluggableUSB().getInterface(&interfaces);
```

In this way the first two endpoint are not being configured, so probably the OS will complain about it but it should work anyway (it works for sure on Linux).

closed 2017-06-19 by facchinm

unicorn79, 2017-06-19:
It works perfectly! Thanks a lot!  IDE 1.8.0 (Windows 7 x64)

oliverdoug, 2017-10-22:
hello unicor79, could you help me with the procedures you performed? I'm having the same problem and tried the tips but I did not succeed.

DanielGibson, 2021-01-05:
It would be very useful to have a standard way to disable the CDC.
A common problem with the CDC is when trying to create a USB Boot Keyboard with an Arduino some devices/systems won't recognize that keyboard because they're confused by the CDC interfaces for example my Aten CS1924 KVM switch is one affected device, https://forum.arduino.cc/index.php?topic=545288.msg3717028#msg3717028 also mentions Windows XP and I can imagine at least some BIOS/UEFI implementations are also affected.
(Side-note: The only reason I'm trying to make an Arduino behave like a keyboard is that my KVM switch doesn't like my new keyboard so I want to put a Pro Micro + USB Host Shield Mini adapter in between them to make it work..)
Another usecase is creating MIDI devices, according to https://forum.arduino.cc/index.php?topic=588469.0

The absolute best solution would be a `Serial.disable();` call that removes those interfaces (and possibly resets the USB device or whatever so the OS re-scans it). That way you could even do something like
```c++
void setup() {
  pinMode(2, INPUT_PULLUP);
  // TODO: would this require a delay()?
  if(digitalRead(2) == LOW) {
    // pin 2 is grounded => enable CDC for debug purposes or whatever
    Serial.begin( 115200 );
    while (!Serial) {}
    Serial.println("Serial console enabled!");
  } else {
    // by default, remove CDC interfaces from the USB device
    Serial.disable();
  }
}
```
But I have no idea how hard that would be; having a `#define` to disable it at compiletime would certainly be better than nothing.

> there is no easy way to get rid of the CDC interface because it's usually the only way to reprogram the board.

I think this is wrong. Even when disabling the CDC interface (with a hacky patch) on my Pro Micro I can still easily reprogram the board directly after grounding the reset pin - I guess it's the same for other devices like the Leonardo?

unicorn79, 2021-01-06:
DanielGibson, thanks for your reply, but Serial.disable(); don't works for my IDE 1.8.0 (exit status 1
'class Serial_' has no member named 'disable').  The task was to leave a functioning bootloader in Leonardo, but remove the com port, that appears after finishing loading Leonardo. This task was successfully solved for my IDE 1.8.0 by the code above from facchinm, thx to him )

DanielGibson, 2021-01-06:
It was just a suggestion of a function that could be implemented to handle this, it doesn't exist (yet?).

unicorn79, 2021-01-06:
DanielGibson, oh yes, of course. It would not be bad if in the future it would be implemented so easily.

DanielGibson, 2021-01-06:
I implemented support for disabling the CDC at compiletime with `-DCDC_DISABLED`: https://github.com/arduino/ArduinoCore-avr/pull/383
Interestingly, the Arduino core for the SAMD architecture already supports this (with the same define)

A runtime switch (like the proposed `Serial.disable()`) would be harder, because so far there doesn't seem to be support for resetting USB or removing devices, so I think I can't implement that myself (at least not without spending a lot more time than I can afford).