arduino/Arduino #6390: HID library does not support callbacks

SteveBenz, 2017-06-17:
HID should support receiving reports back from the host, at least in a minimal fashion.  There's a sketch of what the code might be in the HID library, but it's commented out and not very well fleshed out.  It'd be nice if it was at least good enough to handle keyboard LED messages.

https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/libraries/HID/src/HID.cpp

facchinm, 2017-06-19:
Hi @SteveBenz ,
the builtin HID library (and builtin Mouse and Keyboard) main focus is to show basic functionalities. The effort we made to abstract it from the core gave the chance to write libraries like [HID](https://github.com/NicoHood/HID) by @Nicohood which allows bidirectional communication (https://github.com/NicoHood/HID/blob/master/examples/Keyboard/KeyboardLed/KeyboardLed.ino). However, I like your idea and we could discuss a bit about the user-facing APIs!

SteveBenz, 2017-06-20:
Hi @facchinm , you make a good point.  I'm pretty new to the Arduino scene, but I have to say that one of the frustrations I have with it is that I re-invented a couple of wheels.  This one, for example, because I only just discovered @NicoHood 's work.  It's good to have the basic functionality in the Arduino libraries and then let users extend them.  I wonder, though, when you get something that's as established as Nico's library is now whether the right thing to do is to throw out the Arduino libarary and fully incorporate some large subset of what Nico's got.  Or perhaps just put a big comment in the basic Arduino library directing people to Nico's work if they need to do something more involved.

I'd happily throw this PR out and re-submit one that just adds that comment.  I'm just trying to do what I can to give back, and that comment would surely have done me a lot of good!

facchinm, 2017-06-20:
Well, I totally approve the "adding a comment" strategy. Most users only need basic functionalities and tend to use a lot of libraries so forcing them to include a bigger one (maybe useless for what they need to do) is a no-go. 
Instead, a clear comment in the example sketches which states "For advanced functionalities, try ...bla bla bla... this library" is super interesting. If you want to contribute with a PR I'll be glad to merge it!

SteveBenz, 2017-06-25:
I updated the pull request with a comment that I would have found really helpful had I found it earlier.  I also removed some TODO comments & stub code that I think this conversation shows will not be done.

SteveBenz, 2017-06-26:
Accidentally closed it, apparently.