arduino/Arduino #6395: Digital Potentiometer code example needs delay added

mlfergy, 2017-06-19:
https://www.arduino.cc/en/Tutorial/DigitalPotControl

The code needs a delay to work:

void digitalPotWrite(int address, int value) {
  // take the SS pin low to select the chip:
  digitalWrite(slaveSelectPin, LOW);
**delay(50);**
  //  send in the address and value via SPI:
  SPI.transfer(address);
  SPI.transfer(value);
**delay(50);**
  // take the SS pin high to de-select the chip:
  digitalWrite(slaveSelectPin, HIGH);
}

matthijskooijman, 2017-06-19:
Are you using the exact same chip as mentioned in that tutorial? Did you look up the amount of delay needed in the datasheet? 50ms sounds like a lot more than a typical chip would require.

mlfergy, 2017-06-19:
Exact same chip...  I edited the post.  It actually doesn't work reliably
unless delay is 100 ms

SimonePDA, 2018-04-02:
Fixed the sketch, thanks a lot for your testing.

closed 2018-04-02 by SimonePDA