arduino/Arduino #6397: Limit to visible com Ports

Argus-, 2017-06-19:
I use Nport Administrator to map com ports for remote Moxa port, 48 addition com ports not including local com ports.
When opening the Arduino software I only see 46 com ports and cannot scroll to see anymore, unfortunately none are the com port I'm looking for. If I remove a 8port Moxa port I can see the local com ports.
It seems as if there is a hard limit to the number of com ports the Arduino software will show.

facchinm, 2017-06-20:
Hi @Argus- ,
this is one of the strangest bugs ever reported :slightly_smiling_face: First of all, is the serial port submenu scrollable (or at least shows the little arrow at the bottom)? There is no hard limit on the list itself but probably the menu is rendered as non-scrollable so when you reach the bottom of the screen it stops displaying stuff. Could you provide a screenshot (or a screencast) of the superlong list? Thanks!

Argus-, 2017-06-21:
Hi @facchinm 

I have added a screen shot, it looks like there is more to the list, but as you scroll to the bottom is immediately circles to the top of the list.

The list expands to the full size of my screen but will not scroll to anymore Serial ports and no down arrow to show more either.

[image]

facchinm, 2017-06-22:
Oh I see, the problem really looks related with the absence of the little arrow at the bottom of the list. I'm preparing a pull request so you can test it with that massive number of serial ports :smile:

facchinm, 2017-06-23:
Hi @Argus- ,
could you test PR #6411 in your setup and report if the list is scrollable? Thanks!

Argus-, 2017-06-23:
Hi @facchinm 

It works, but sometimes it has an issue displaying the full list or scrolling full list, I've
 re-added 8 addition ports and so I have approximately 55 total mapped com ports.

[image]
[image]
[image]

facchinm, 2017-06-26:
Argh, I think I understand what the problem is... When the list gets updated the arrows are being added anyway, despite the fact that they are already there. I'm modifying the PR a bit to try to fix this problem. Thanks!

PR #6411 (2017-08-01): Add scroller for serial port list menu
Fixes #6397
commit 506ebd80 (2017-08-01): add scroller for serial port list menu

Fixes #6397
closed 2017-08-01 by facchinm