arduino/Arduino #6398: IDE crashes without error when choosing Sketch->Include Library->Add .ZIP Library...

PhilipKyleBoone, 2017-06-19:
I recently updated Windows 10 with the Creators Update. Since then, the Arduino IDE crashes without error (All open IDE Windows get closed without warning) when I select the Menu "Sketch->Include Library->Add .ZIP Library..."

Hope you get this fixed soon.

am using Arduino IDE 1.8.3

matthijskooijman, 2017-06-20:
Does this happen with any .zip library or only specific ones? Can you share a .zip file that causes this error for you? Could you try opening Arduino from a terminal window to see if any error is shown there (start arduino_debug.exe from cmd.exe)?

facchinm, 2017-06-20:
@PhilipKyleBoone if you could attach the zip here it would be super useful, thanks!

PhilipKyleBoone, 2017-06-20:
Hi Guys,

The IDE crashes as soon as I click the Menu "Add .ZIP Library" i.e. the Open File Dialog does not get shown.

I Haven't tried re-installing the IDE yet, so I will try that this evening and let you know.

facchinm, 2017-06-20:
Does it also crash if you try to open a file via `File/Open` menu? Before reinstalling (which could solve the problem) try launching the IDE with `arduino_debug.exe` and paste here the output so we can debug the problem for everyone :wink: Thanks a lot!

PhilipKyleBoone, 2017-06-20:
Hey,
so, File/Open works fine.
I'm attaching the error dump log to this post.
It may help to know, that I have Visual Studio 2015 and 2017 Enterprise versions installed. I also have the vMicro Plug-In installed. But like I said, the Problem just started right after the 
latest Windows Update... the Creator Update.
Thanks for looking into this Problem.

[hs_err_pid15536.zip](https://github.com/arduino/Arduino/files/1089237/hs_err_pid15536.zip)

facchinm, 2017-06-21:
Thanks a lot for the dump! The bug (in Java JVM) has just been fixed (see https://bugs.openjdk.java.net/browse/JDK-8179014) and will probably be included in the upcoming release (u141) . It looks related with the so-called [GodMode](https://www.cnet.com/how-to/activate-godmode-in-windows-10/) which listing results in a NULL pointer on Win10 Creators. Could you try removing that folder or any program which might have created it and check if the menu works again? Thanks a lot!

facchinm, 2017-06-22:
Very well :smile: I'm leaving the issue open as a reminder to update the bundled Java as soon as u141 is out. Thanks you!

DrJFM, 2017-07-19:
Same issue. Same Godode Folder, deleted it and all fine!  Win 10 Creator, 1.8.3.  Glad I found this!

facchinm, 2017-07-31:
I've just updated jre to 1.8.0_u144 on our build server, so latest nightly should contain the "God mode" fix. Could anyone test[ this build ](https://www.arduino.cc/download.php?f=/arduino-nightly-windows.zip) with God mode active so we can close the issue with confidence :wink: ?