arduino/Arduino #6399: necesito actualización

storm2pac, 2017-06-20:
el entorno de programación arduino podría ser mucho mas fácil e intuitivo si dentro del entorno existe una herramienta de "flowcharts" o diagrama de flujos. Programadores con un mínimo conocimiento se adentran al mundo arduino de una manera mas fácil.

facchinm, 2017-06-20:
Hi @storm2pac ,
what are you referring about? Can you provide a link (or write in english so everyone can contribute to the discussion)? Thanks!

storm2pac, 2017-06-20:
Hi facchinm, I'm using arduino and I came up with an idea "maybe arduino with flowcharts will be easier". The very famous proteus program uses them in "visual designer".
I may not be driving in the right place
A greetings   :argentina: :smile:

facchinm, 2017-06-21:
HI @ghost ,
there are plenty of visual programming tools based on Arduino but we officially support none; they are mostly targeted to educational markets than professional but they can also be used to produce really complex stuff. Take a look at https://ardublockly.embeddedlog.com/ and http://technologiescollege.github.io/Blockly-at-rduino/ for some examples (or google for "viasual programming arduino" :wink:  ) Thanks!

closed 2017-06-21 by facchinm