arduino/Arduino #6400: Serial Port dropdown - simpler handling multiple Arduinos / serial ports

seeers, 2017-06-20:
I have a suggestion to improove work with several Arduinos: It is very cumbersome to change the serial port via the menu. A small dropdown list with the available serial ports next to the Serial Monitor button would be a great relief.

LowPowerLab, 2017-11-02:
I suggest having an optional toolbar with buttons that represent the available serial ports (ex. labeled "COMx"), with the active one highlighted. As ports become un/available the port-buttons would reflect that.
This same concept would be great for different target boards, say have a toolbar with buttons to select the last few target boards that were used, to allow easy switching between them, rather than through the menu.

feikname, 2017-11-02:
An keyboard shortcut for selecting the only available port might be very useful too.

LowPowerLab, 2017-11-02:
In my case multiple ports are always active, and usually I keep exchanging a target board which enumerates a new serial port each time, so clicking UPLOAD fails because the old port is now gone and I have to pick the new one from the Menu. In Arduino 1.0.6 there used to be a serial picker popup/dialog to choose a new serial port, which had a dropdown with the first available serial/COM port automatically selected so all I had to do was hit ENTER to activate that newly enumerated serial port. That dialog is now gone, instead upload just fails and the avr_dude ser_port() exception is shown in the status console.

LowPowerLab, 2017-11-02:
I think this is because in **SerialUploader.java** all exceptions are repackaged as RunnerException.

```
    //SerialUploader.java
    boolean uploadResult;
    try {
      String pattern = prefs.getOrExcept("upload.pattern");
      String[] cmd = StringReplacer.formatAndSplit(pattern, prefs, true);
      uploadResult = executeUploadCommand(cmd);
    } catch (RunnerException e) {
      throw e;
    } catch (Exception e) {
      throw new RunnerException(e);
    } finally {
      BaseNoGui.getDiscoveryManager().getSerialDiscoverer().pausePolling(false);
    }
```

In **Editor.java** the Upload handler there is actually a catch(SerialNotFoundException) which would invoke the serialPrompt which I mentioned was working in 1.0.6. But here if the serialUpload code above eats all serial exceptions and packages them as RunnerException, this would obviously explain why that dialog never has a chance to work anymore:

```
  //Editor.java
  // DAM: in Arduino, this is upload
  class DefaultExportHandler implements Runnable {
    public void run() {

      try {
        removeAllLineHighlights();
        if (serialMonitor != null) {
          serialMonitor.suspend();
        }
        if (serialPlotter != null) {
          serialPlotter.suspend();
        }

        uploading = true;

        boolean success = sketchController.exportApplet(false);
        if (success) {
          statusNotice(tr("Done uploading."));
        }
      } catch (SerialNotFoundException e) {
        if (portMenu.getItemCount() == 0) statusError(e);
        else if (serialPrompt()) run();
        else statusNotice(tr("Upload canceled."));
      } catch (PreferencesMapException e) {
        statusError(I18n.format(
                    tr("Error while uploading: missing '{0}' configuration parameter"),
                    e.getMessage()));
      } catch (RunnerException e) {
        //statusError("Error during upload.");
        //e.printStackTrace();
        status.unprogress();
        statusError(e);
      } catch (Exception e) {
        e.printStackTrace();
      } finally {
        populatePortMenu();
        avoidMultipleOperations = false;
      }
```