arduino/Arduino #6402: Weird errors when sketch name and directory name differ in capitalization only...

WestfW, 2017-06-20:
If you have a sketch in a directory whose name is different only in capitalization, the Arduino IDE will get confused about what type of file it is pre-processing.

See https://github.com/WestfW/OptiLoader/pull/12

Duplicate by:

1. Download optiLoader .zip from https://github.com/WestfW/OptiLoader and uncompress.

1. This gives you a directory OptiLoader-master, which the Arduino IDE won't like.  Rename it to just "OptiLoader" - now the IDE will NOT complain about needing to "move" the sketch into a matching directory. (Note that the capitalization is still different.)

1. Try to compile ("Verify") - the IDE will apparently attempt to do .cpp-style pre-processing on the optiLoader.h file, resulting in an error:

```
/Applications/arduino/Arduino-1.8.2.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware -hardware /Applications/arduino/Arduino-1.8.2.app/Contents/Java/portable/packages -tools /Applications/arduino/Arduino-1.8.2.app/Contents/Java/tools-builder -tools /Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/tools/avr -tools /Applications/arduino/Arduino-1.8.2.app/Contents/Java/portable/packages -built-in-libraries /Applications/arduino/Arduino-1.8.2.app/Contents/Java/libraries -libraries /Applications/arduino/Arduino-1.8.2.app/Contents/Java/portable/sketchbook/libraries -fqbn=arduino:avr:uno -ide-version=10802 -build-path /var/folders/jz/5yb8f2hr8xjcpf0059bsfz4r0000gn/T/arduino_build_573999 -warnings=none -build-cache /var/folders/jz/5yb8f2hr8xjcpf0059bsfz4r0000gn/T/arduino_cache_401912 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/tools/avr -verbose /Volumes/Downloads/Downloads/OptiLoader/optiLoader.h

/Applications/arduino/Arduino-1.8.2.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware -hardware /Applications/arduino/Arduino-1.8.2.app/Contents/Java/portable/packages -tools /Applications/arduino/Arduino-1.8.2.app/Contents/Java/tools-builder -tools /Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/tools/avr -tools /Applications/arduino/Arduino-1.8.2.app/Contents/Java/portable/packages -built-in-libraries /Applications/arduino/Arduino-1.8.2.app/Contents/Java/libraries -libraries /Applications/arduino/Arduino-1.8.2.app/Contents/Java/portable/sketchbook/libraries -fqbn=arduino:avr:uno -ide-version=10802 -build-path /var/folders/jz/5yb8f2hr8xjcpf0059bsfz4r0000gn/T/arduino_build_573999 -warnings=none -build-cache /var/folders/jz/5yb8f2hr8xjcpf0059bsfz4r0000gn/T/arduino_cache_401912 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/tools/avr -verbose /Volumes/Downloads/Downloads/OptiLoader/optiLoader.h

Using board 'uno' from platform in folder: /Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/arduino/avr

Using core 'arduino' from platform in folder: /Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/arduino/avr

Detecting libraries used...
"/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/jz/5yb8f2hr8xjcpf0059bsfz4r0000gn/T/arduino_build_573999/sketch/optiLoader.h.cpp" -o "/dev/null"

"/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/arduino/Arduino-1.8.2.app/Contents/Java/hardware/arduino/avr/variants/standard" "/var/folders/jz/5yb8f2hr8xjcpf0059bsfz4r0000gn/T/arduino_build_573999/sketch/optiLoader.h.cpp" -o "/var/folders/jz/5yb8f2hr8xjcpf0059bsfz4r0000gn/T/arduino_build_573999/preproc/ctags_target_for_gcc_minus_e.cpp"

/Volumes/Downloads/Downloads/OptiLoader/optiLoader.ino:85:24: fatal error: optiLoader.h: No such file or directory
 #include "optiLoader.h"
```

Easiest fix is probably to fix the check where the IDE (incorrectly) thinks that the sketch name and directory name match...

facchinm, 2017-06-21:
The problem looks related with a partial move when the new folder gets created. Bugfix ongoing :wink: Thanks for the report!

facchinm, 2017-06-23:
Hi @WestfW ,
could you please test-run PR #6404 and report if it works correctly in your setup? Thanks!

WestfW, 2017-06-24:
No; still behaves the same on MacOS version.
note when I say 'rename' in the initial bug report, I'm talking about a "rename" outside of the arduino IDE itself (using Finder on MacOSX or File Explorer on Windows.)  The IDE never does any renaming or "moving" of the sketch (unless you're talking about the <temp-buildDir>/sketch directory, which strangely gets a "optiLoader.h.cpp" but NOT an "optiLoader.ino.cpp"
(When the original sketch folder name is correctly capitalized, I get "optiLoader.ino.cpp" and "optiLoader.h" in the <build>/sketch/ directory (as expected.)

Note that in this case, the problematic .h file has the same filename root as the sketch name.

facchinm, 2017-06-26:
OK, so it is related with case-insensitive filesystems and I fixed another unrelated bug :smile: I'll check on OSX and report back, but I think this is triggered by the [builder](https://github.com/arduino/arduino-builder) and not by Java IDE.

commit bd04a40a (2017-10-16): Moving a sketch to another folder should move all files

Fixes #6402
commit 258d6d5a (2017-10-23): Moving a sketch to another folder should move all files

Fixes #6402
commit 788884fb (2017-10-30): Moving a sketch to another folder should move all files

Fixes #6402
commit ff308dbf (2017-11-10): Moving a sketch to another folder should move all files

Fixes #6402
closed 2018-05-02 by facchinm