arduino/Arduino #6407: Builder crashes on second run

MicioMax, 2017-06-21:
I've a couple of big sketches (custom board and platform files...) which crashes the builder on second run. I have to close and restart the IDE to rebuild it.
Here the java crash log... please tell me if you need more :

panic: runtime error: index out of range

goroutine 1 [running]:
panic(0x556bc0, 0xc420010140)
	/home/jenkins/go/src/runtime/panic.go:500 +0x1a1
arduino.cc/builder.findIncludesUntilDone(0xc420034c00, 0xc4203823f0, 0x556f40, 0xc42039b4a0, 0xc4202840d4, 0x10, 0x0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/container_find_includes.go:321 +0x138f
arduino.cc/builder.(*ContainerFindIncludes).Run(0x64d248, 0xc420034c00, 0x41bc3c3c, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/container_find_includes.go:149 +0x5e4
arduino.cc/builder.runCommands(0xc420034c00, 0xc420473ae8, 0x22, 0x22, 0x1, 0x0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:191 +0xcd
arduino.cc/builder.(*Builder).Run(0xc420473d38, 0xc420034c00, 0xc420045d88, 0x562240)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:124 +0xb9c
arduino.cc/builder.RunBuilder(0xc420034c00, 0xc420045d88, 0x64d248)
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/builder/builder.go:222 +0x35
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross/src/arduino.cc/arduino-builder/main.go:338 +0x99c
/home/massimo/Documenti/Elettronica/ARDUINO/arduino-1.8.3/arduino-builder ha restituito 2
Errore durante la compilazione per la scheda Fishino32.

matthijskooijman, 2017-06-22:
I think this is the same as https://github.com/arduino/arduino-builder/issues/230. What board are you compiling for? Could you see if the arduino-builder from https://github.com/arduino/arduino-builder/issues/230 fixes this?

matthijskooijman, 2017-06-22:
Correction, the build from https://github.com/arduino/arduino-builder/pull/236

MicioMax, 2017-06-22:
Yep... the crash is away.
For the meanwhile, is it possible to modify the platform.txt to make the unpatched IDE work ?
I've got many people using my package, and most of them don't know how to patch it...

MicioMax, 2017-06-22:
mhhhh.... it shows no error, but it has problems; here the last part of the output, followed by last part of the output of my own IDE (self coded), which uses same platform data; you can see that sketch size is ONE THIRD in arduino ide, and it doesn't even start on my board :
ARDUINO IDE OUTPUT
`Uso la libreria FishinoDebug alla versione 6.0.0 nella cartella: /home/massimo/sketchbook/libraries/FishinoDebug 
Uso la libreria Wire nella cartella: /home/massimo/.arduino15/packages/fishino/hardware/pic32/6.0.0/libraries/Wire (legacy)
Uso la libreria Timer alla versione 1.0.0 nella cartella: /home/massimo/.arduino15/packages/fishino/hardware/pic32/6.0.0/libraries/Timer 
Uso la libreria SPI nella cartella: /home/massimo/.arduino15/packages/fishino/hardware/pic32/6.0.0/libraries/SPI (legacy)
Uso la libreria I2S alla versione 5.1.0 nella cartella: /home/massimo/.arduino15/packages/fishino/hardware/pic32/6.0.0/libraries/I2S 
Uso la libreria FishinoStream alla versione 6.0.0 nella cartella: /home/massimo/sketchbook/libraries/FishinoStream 
Uso la libreria FishinoAudioCodec alla versione 6.0.0 nella cartella: /home/massimo/sketchbook/libraries/FishinoAudioCodec 
Uso la libreria FishinoAudioReader alla versione 6.0.0 nella cartella: /home/massimo/sketchbook/libraries/FishinoAudioReader 
Uso la libreria FishinoPlayer alla versione 6.0.0 nella cartella: /home/massimo/sketchbook/libraries/FishinoPlayer 
Uso la libreria FishinoFileStream alla versione 6.0.0 nella cartella: /home/massimo/sketchbook/libraries/FishinoFileStream 
Uso la libreria FishinoSdFat alla versione 6.0.0 nella cartella: /home/massimo/sketchbook/libraries/FishinoSdFat 
Lo sketch usa 118168 byte (22%) dello spazio disponibile per i programmi. Il massimo è 524288 byte.
Le variabili globali usano 6792 byte (5%) di memoria dinamica, lasciando altri 124280 byte liberi per le variabili locali. Il massimo è 131072 byte.`

FISHIDE output :
`Creazione archivio '_SKETCH_.ar' : [ FishinoPlayerTest.ino.cpp ]
Linking di FishinoPlayerTest.elf : [ _CORE.ar LIB_FishinoDebug.ar LIB_Wire.ar LIB_Timer.ar LIB_SPI.ar LIB_I2S.ar LIB_FishinoStream.ar LIB_FishinoAudioCodec.ar LIB_FishinoAudioReader.ar LIB_FishinoPlayer.ar LIB_FishinoSdFat.ar LIB_FishinoFileStream.ar _SKETCH_.ar ]
Creazione del file HEX 'FishinoPlayerTest.hex' : [ FishinoPlayerTest.elf ]
Getting size for 'FishinoPlayerTest.hex'
Sketch output sizes:
  CODE   :     316588 bytes
  DATA   :       6768 bytes
  EEPROM :       3948 bytes
`

matthijskooijman, 2017-06-22:
For an example to fix this in platform.txt, see https://github.com/chipKIT32/chipKIT-core/pull/356
Essentially you're just inlining the compiler.c.flags var into the compiler.cpp.flags var.

As for your other error, I suspect it is a different and unrelated problem. The difference in code size is remarkable, so that would be a good place to start debugging (I have [some scripts on github](https://github.com/matthijskooijman/scripts/tree/master/Arduino) that can help figuring out what's inside and what's the difference). Comparing the full list of gcc commands between both IDEs might also be useful (perhaps different board options or defines are passed on the commandline? Different optimization options?).

MicioMax, 2017-06-22:
Hi matthijskooijman,
yep, I solved the crash adding compiler.preproc.flags without the -MMD and it builds now.
The size problem is still there, I guess it's a problemi with the linker, I'll investigate it.
In my IDE I use a totally different approach, making all the .ar first and then linking them together.
Probably IDE's linker is missing some library code... the weird stuff is that it doesn't complain at all.
Smaller sketches do work indeed.

MicioMax, 2017-06-22:
(options are exactly the same on both IDEs. Mine has provisions for changing the optimizations, but I checked the compiler commands and it's exactly the same, just include's order change. Linker command is quite different)

per1234, 2017-08-03:
@MicioMax any news on the code size difference? Can this issue be closed now or is there still an outstanding issue with the Arduino IDE?

closed 2020-12-24 by per1234