arduino/Arduino #6408: the find window not always on top of the main window

czghzc, 2017-06-22:
the find window not always on top of the main window

facchinm, 2017-06-22:
As far as I can tell, the "find" window always appears at the center of the active IDE window. 
Are you experiencing something different? Can you provide a screenshot? Thanks!

czghzc, 2017-06-22:
[image]

facchinm, 2017-06-23:
Mmmmh, it looks like a problem with Windows' window manager  (and probably Java). I'll try investigating before updating to the next Java version.

czghzc, 2017-06-24:
My system is Win7-32Bit.
My previous version of Java is 1.7.0_80-b15, there is such a problem.
Then I updated the Java version to 1.8.0_131-b11, the problem is still not resolved.
If you have no problem with the test, it may be my system.

czghzc, 2017-10-07:
[image]

I'm sorry, but now I understand your reply, it is ashamed.
I tested it, and now the find window is above all windows, but it finds another problem.
If there are other windows in the system, such as a browser, the find window will also be on top of the browser window, which will interfere with the operation of the browser window, and we hope that the find window is just above the Arduino main window.

kenneth558, 2018-02-28:
Agreed that the find window drops behind main window.  Linux Mint 18, IDE 1.8.5, but has always been the case in IDE on Linux

facchinm, 2018-03-02:
Would you mind testing https://github.com/arduino/Arduino/pull/6603 (or the latest beta)? The wrong behaviour should be gone for good.

czghzc, 2018-03-09:
[image]
The beginning is good, there is no problem, but after clicking the window's title bar with a mouse is not normal, the search window goes to the back of the main window.

When programming in C++, setting the search window to the child window of the main window can achieve the desired effect. However, I will not use Java. I do not know if Java has the same operation method.