arduino/Arduino #6409: Arduino WiFi Shield with IDE 1.8.1 gives wrong values for multiple byte TCP read in server mode

Sudeshna19, 2017-06-22:
I have Mega 2560 connected to WiFi Shield (configured as server). I am using IDE 1.8.1. In my code I want to receive say 2 bytes. The client sends 1 byte at a time. Say client sends b first and then A ideally server should read bA after the second byte sent from client. However, I notice in IDE 1.8.1 the values get corrupted. I have attached the code, also pasting it below:
`#include <SPI.h>
#include<WiFi.h>

int configureSuccess = WL_IDLE_STATUS;
char ssid[] = "xyz";
char pass[] = "*******";
WiFiServer server(20000);

static uint8_t alreadyConnected = 0;

void setup() {
Serial.begin(9600);
Serial1.begin(28800);
Serial.print("Attempting to connect to SSID: ");
Serial.println(ssid);
while ( configureSuccess != WL_CONNECTED) {
configureSuccess = WiFi.begin(ssid, pass);
// delay 10000
unsigned long startMillis = millis();
while(millis()- startMillis < 10000)
{
};
}
server.begin();

  for(int trialcount=0;trialcount<5;trialcount++)
  {
      if(Serial)
      {
          break;
      }else
      {
          unsigned long startMillis  = millis();
    while(millis()- startMillis < 1000)
                    {
                    };
      }
  }
  
  if (configureSuccess==WL_CONNECTED)
   { 
     IPAddress ip = WiFi.localIP();
     Serial.print("<<<IP address: ");
     Serial.print(ip);  
     Serial.println(">>>");            
  }
 else
 {
     Serial.println("<<< IP address :Failed to configure. >>>");
 }
}

void loop() {
// wait for a new client:
int avlBytes = 0;
int sizeReceived = 0;
int data2[2];

Serial1.println("loop");
WiFiClient client1 = server.available();
if (client1 == true){
if(!alreadyConnected){
client1.flush();
alreadyConnected++;
Serial1.println("Flushing");
}
avlBytes = client1.available();
if(avlBytes >= 2){
data2[0] = client1.read();
data2[1] = client1.read();
Serial1.println("data_s");
[wifishield1_6_13.zip](https://github.com/arduino/Arduino/files/1093719/wifishield1_6_13.zip)

Serial1.println(data2[0]);
Serial1.println(data2[1]);
}
else{
Serial1.println("rcvd");
Serial1.println(avlBytes);
}
}
else {
alreadyConnected = 0;
Serial1.println("No client");
}
unsigned long startMillis = millis();
while(millis()- startMillis < 1000)
{
};
}`

If client send b first then rcvd 1 is printed. If client then sends A then data_s -1 -1 is printed. If I try the same code in IDE 1.6.13 data_s 98 65 is printed which is correct.

However, in both IDE 1.8.1 and IDE 1.6.13 if I use client1.read(data2, 2) instead of two back-to-back read() the Serial1 print would be data_s 98 0. The second byte can be anything random, not always 0.

I also tried out the same code in MKR1000 and IDE 1.8.1. I used size_rcvd = client1.read(data2, 2) instead of two read(). I saw the prints as data_s 65 0 even though the size_rcvd = 2. It was always neglecting the previously sent byte.

wifishield1_6_13.zip

facchinm, 2017-06-22:
Hi @Sudeshna19 ,
the question should be moved, either to the [forum](https://forum.arduino.cc) or to the relevant repos ([WiFi](https://github.com/arduino-libraries/WiFi) or [WiFi101](https://github.com/arduino-libraries/WiFi101).
The problem is quite peculiar since, between 1.6.13 and 1.8.1 the WiFi library was not updated, so I think the problem could be in the sketch itself (for example, the line ```if (client1 == true)``` makes no sense). Try posting your problem on the forum and you'll surely get some precious advice! Thanks.

closed 2017-06-22 by facchinm

per1234, 2017-06-23:
Just to link everything together, I wanted to note that the issue was reposted at:
https://github.com/arduino-libraries/WiFi/issues/30
and
https://github.com/arduino-libraries/WiFi101/issues/183

Sudeshna19, 2017-06-23:
Yeah.... I created arduino-libraries/WiFi#30 for the WiFiclient::read(*dataBuf, datasize) giving incorrect results than WiFiClient::read() (single byte read) in WiFi library (IDE 1.8.1).

I also created arduino-libraries/WiFi101#183 for WiFi101 issue: WiFiClient::read(*dataBuf, datasize) giving incorrect results when used on MKR1000.