arduino/Arduino #6410: SSH Upload using proxy regardless of preference settings on Win 10 (1.8.3)

rjdza, 2017-06-22:
If a system proxy is set on Windows 10, the SSH upload will fail.  This happens even when I have set the Arduino proxy settings to "no proxy" in the network tab.

Upload works as expected if the system proxy is turned off.

The proxy server shows no connections, so I'm not sure what is happening there, but tcpdump on the proxy, and wireshark on the desktop both show connections from the desktop to the proxy on the proxy port.

Here is the Arduino error - it basically says it cannot establish a socket, which leads me to believe that it is, for some reason, trying to SSH into the proxy server on the proxy port:

`Sketch uses 3800 bytes (11%) of program storage space. Maximum is 32256 bytes.
Global variables use 314 bytes (15%) of dynamic memory, leaving 1734 bytes for local variables. Maximum is 2048 bytes.
processing.app.debug.RunnerException
	at cc.arduino.packages.uploaders.SSHUploader.uploadUsingPreferences(SSHUploader.java:145)
	at cc.arduino.UploaderUtils.upload(UploaderUtils.java:78)
	at processing.app.SketchController.upload(SketchController.java:713)
	at processing.app.SketchController.exportApplet(SketchController.java:686)
	at processing.app.Editor$DefaultExportHandler.run(Editor.java:2149)
	at java.lang.Thread.run(Thread.java:745)
Caused by: com.jcraft.jsch.JSchException: timeout: socket is not established
	at com.jcraft.jsch.Util.createSocket(Util.java:389)
	at com.jcraft.jsch.Session.connect(Session.java:215)
	at cc.arduino.packages.uploaders.SSHUploader.uploadUsingPreferences(SSHUploader.java:117)
	... 5 more
`

facchinm, 2017-06-22:
Hi @rjdza ,
you are right, the SSH upload does not take into account any kind of proxy (system wide or user specified). This happens because in most proxy configurations port 22 and port 443 are forwarded transparently since they are already encrypted and the redirect would appear as a man-in-the-middle attack. Could you give some more insights about how your setup is configured and if you are able to SSH into the remote machine using any other SSH client (like putty)? Thanks!

rjdza, 2017-06-22:
There as been a miscommunication on my side.  If the system proxy is set, the IDE will try to connect to the proxy server instead of the SSH server, and it connects on the proxy port.  If I disable the system proxy, everything works just fine.

I am connecting to an  [Onion Omega2 with the Arduino dock](https://onion.io/store/arduino-dock-r2/).  

I haven't tried Putty (although I can't see why it would make a difference), but OpenSSH works just fine from Cygwin and Linux.  WinSCP, rsync and a custom client I wrote in python also work.  This is not a SSH issue, pretty sure of that. :)  I've tested from 3 desktops and to 3 Omegas.

I may not be expressing myself clearly, I've been up all night.  Please let me know if I'm not making sense...

facchinm, 2017-06-22:
Mmmmh, this is super strange. As per [Jsch documentation](https://epaul.github.io/jsch-documentation/simple.javadoc/com/jcraft/jsch/Session.html#setProxy-com.jcraft.jsch.Proxy-) the proxy should be explicitly set if we want to use it, otherwise the communication should be plain TCP (on the correct IP and port, I hope).
Something in the system proxy should be redirecting the connection anyway, despite our settings. 
Which OS / proxy combination are you using exactly? 

One other chance is that mdns is getting confused; which address is shown on the Port menu? The board one or the proxy's ?

rjdza, 2017-06-22:
Desktops are Mac OSx, Linux and Windows 10, servers are all Linux & BSD - combination of Ubuntu and CentOS.  The Proxy is CentOS running squid, but if I point my proxy to the firewall (BSD) I get the same problem.  

I don't think the system proxy is doing anything to the connection (pretty sure it's not, actually) because I have (many, many) other services I connect to that work just fine and don't go near the proxy.

I'm not sure about mdns - I don't have a great amount of experience with it.  I can say, though, that the IP address reported in the tools/port section reports the correct hostname / ip information. that is to say it points to the Omega, not to the proxy.

facchinm, 2017-06-23:
Argh, if the ip is correct I really can't see where the problem could be. 
To sum up:
- if no system proxy is set, the upload works correctly
- if system proxy is on and IDE is configured not to use it, upload fails
- if system proxy is on and IDE is configured to use it, upload fails too

I'll investigate a bit with a squid local proxy and I'll let you know if I discover something :slightly_smiling_face: 
Thanks!

rjdza, 2017-06-23:
Your summation is correct.

Please let me know if there is anything I can do to help, and thanks to you :)