arduino/Arduino #6416: overwrite existing sketch not possible

tofrnr, 2017-06-23:
overwrite an existing sketch not possible - given there is already a sketch test.ino in a folder test, 
project\test\test.ino
and from the upper directory project\  I want to save a current opened sketch (perhaps named "test_2" ) now as "test" in order to overwrite the former one,
then the IDE instead creates a new subdir inside the existing dir project\test  with a new file test.ino inside 
project\test\test\test.ino
and the existing file
project\test\test.ino
keeps unchanged and undeleted.

That is weird, one must expect when saving a new file under the name of an existing one that simply overwriting will be  possible, after the usual check-up window 
"sketch file .... already exists -  do you really want to overwrite? 
yes - no - cancel".

(searching for "overwrite existing sketch" does not show up matching topics)

tofrnr, 2017-06-27:
why has this not been tagged as " Component: IDE / Type: Bug " meanwhile?

matthijskooijman, 2017-06-27:
I guess they sometimes want to get some things finished without being interrupted by all the issues that get reported in the meanwhile :-p I'll add the labels, and I was considering to have a look at this issue, since it might be related to some cleanups I did a while ago.

matthijskooijman, 2017-06-27:
I'm not sure yet whether this is a bug or a missing feature, though.

tofrnr, 2017-06-27:
thank you for looking at it!

johnmbarrett, 2017-12-06:
Ran into something similar to this today, but the behaviour does not match the OP's description.  Instead, if you try to overwrite an existing .ino file, it turns that file into a folder, and saves the new file inside that folder, appending .ino to the original name.

Steps to reproduce:

1). Create a new sketch and save it as "test"
2). Close that sketch and create another new one
3). Select File->Save As
4). Go into the "test" folder in your Arduino code folder
5). Select "test.ino"
6). Press Save
7). Press Yes when prompted to overwrite

Expected result: "test.ino" is overwritten with the new sketch.

Actual result: a new folder inside "test" called "test.ino", which contains a file "test.ino.ino".  You can repeat this process as many times as you like, getting files and folders with progressively more ".ino"s.

I was running Arduino 1.6.9 when I ran into this but it was still there when I updated to 1.8.5.  Running Windows 7 x64, FWIW

facchinm, 2017-12-11:
Hi @johnmbarrett ,
I pushed a fix for this problem in Beta branch and it's now available on https://www.arduino.cc/en/Main/Software#beta. Can you test it and report if it works as intended? Thank you!

closed 2018-05-10 by facchinm