arduino/Arduino #6417: Arduino programming error

Springbok12345, 2017-06-23:
This is my first Arduino programme and it doesn't work. Can anybody please help me because I'm just starting to learn about programming.

1 int switchState = 0;

2 void setup(){
  3 pinMode(3,OUTPUT);
  4 pinMode(4,OUTPUT);
  5 pinMode(5,OUTPUT);
  6 pinMode(2,INPUT);
  7 }
  8 void loop(){
  9 switchState = digitalRead(2); 
  10 if (switchState == LOW);  
  11 then{
  12 digitalWrite(3, HIGH); 
  13 digitalWrite(4, LOW); 
  14 digitalWrite(5, LOW); 
  15 }
  16 else;{ 
  17 digitalWrite(4, HIGH);
  18 digitalWrite(5, LOW);
  19 delay(250) 
  20 }  
Every time I verify, it tells me that there is an error in line eight.
Could anybody help, please? This is starting to get really frustrating.

matthijskooijman, 2017-06-23:
First off - this is not the right place to ask for help with your own code. Did you read the [guidelines for contributing](https://github.com/arduino/Arduino/blob/master/CONTRIBUTING.md) that were linked above the new issue form? I'll be closing this issue for that reason.

Looking at your code anyway, I see you messed up the syntax for the "if" on line 10/11. There is no such thing as a "then" keyword, and there should not be a ; after the if (...) and the else. Interestingly when I try your code, I get an error on line 10, not 8.

closed 2017-06-23 by matthijskooijman

Springbok12345, 2017-06-23:
Hi,

This is not my code. It is the code from the book I received with my
project.

Springbok12345, 2017-06-23:
Hi Mattjis,

Thank you very much for your guidance. I bought this arduino kit for my son
but he got stuck on the first project because the code doesn't work.

We will read a bit more to find where to get help.

Kind regards
Annette van Zyl

cmaglie, 2017-06-23:
The forum is a good place to start:

https://forum.arduino.cc/

Springbok12345, 2017-06-26:
Thank you very much!