arduino/Arduino #6425: Add column-mode editing to the IDE's text editor

TheBetaFruit, 2017-06-24:
It would be awesome to have a column-mode for the editor. This makes it really easy to replace whole columns of content at once.

Like in Visual Studio or Notepad ++, it could be activated using the Alt+Shift+ArrowKey Combination.

facchinm, 2017-06-26:
This should be first implemented by RSyntaxTextArea (issue here https://github.com/bobbylight/RSyntaxTextArea/issues/90), feel free to comment on that issue to gain traction and speedup the implementation :wink:

per1234, 2017-07-01:
It should be noted that this feature was previously available in the Arduino IDE before the switch to RSyntaxTextArea in Arduino IDE 1.6.5 (see https://forum.arduino.cc/?topic=333659).

CyberSwatTeam990000, 2021-03-02:
Is this ever coming? It's a pretty usual thing. I wanna comment out a whole method, I have to go line to line and double tap, what is this the 1970's lol

per1234, 2021-03-02:
Great news everyone! This requested functionality is now available in [Arduino IDE 2.x](https://github.com/arduino/arduino-ide). By default, it uses the same standard keyboard shortcut as VS Code and Notepad++.

It's still in the beta development phase, but if you like, you're welcome to try it out:
https://www.arduino.cc/en/software#experimental-software

closed 2021-03-02 by per1234