arduino/Arduino #6426: delay() and "void init( void )" can't be used  in the same sketch

MitchiLaser, 2017-06-25:
Using the _delay()_ funktion and an self-written funktion called "_void init ( void )_" is not possible and causes the following error:

wiring.c.o (symbol from plugin): In function `__vector_16':
(.text+0x0): multiple definition of `init'
sketch/sketch_jun25a.ino.cpp.o (symbol from plugin):(.text+0x0): first defined here
collect2: error: ld returned 1 exit status
exit status 1
Fehler beim Kompilieren für das Board Arduino Nano.

PaoloP74, 2017-06-25:
Funcion init() is defined here: https://github.com/arduino/Arduino/blob/master/hardware/arduino/avr/cores/arduino/Arduino.h#L128

MitchiLaser, 2017-06-25:
But there is still no possibility to use a function called _void init(void)_ inside an arduino Sketch

facchinm, 2017-06-26:
Hi @MitchiLaser ,
`init()` function is reserved, just like `main()` . In your sketch you should use less generic names or wrap everything in a different namespace (or a class, as you prefer).

closed 2017-06-26 by facchinm