arduino/Arduino #6431: Library Manager addition

justind000, 2017-06-27:
Please add the following repository to the library manager listing.

https://github.com/u-fire/ECSalinity

cmaglie, 2017-06-29:
I've added the library to the indexer but there is a problem:

```
JOB 1023 - 2017/06/29 16:25:40 Scraping https://github.com/u-fire/ECSalinity.git
JOB 1023 - 2017/06/29 16:25:44 Checking out tag: v1.0.0
JOB 1023 - 2017/06/29 16:25:44 Error syncing library: Error generating library from repo: 'author' and 'maintainer' fields must be defined
```

you must define `author` and `maintainer` in `library.properties`.
To fix this:

1) add the missing fields in library.properties
2) commit & push
3) remove and redo the `1.0.0` tag.

once done the library will be automatically published within an hour.

closed 2017-06-29 by cmaglie

njh, 2021-08-18:
@justind000 it looks like this library has now been deleted?
Do you have any information about why?

Thanks!

justind000, 2021-08-19:
Hello, I submitted some newer libraries awhile ago and requested this be removed. Maybe it didn't get removed?

https://github.com/arduino/library-registry/commit/24f2b422d3601c325693a518856a285e43f57872

per1234, 2021-08-19:
>Maybe it didn't get removed?

That is correct. We received your removal request and I started the process at https://github.com/arduino/library-registry/pull/221, but the index maintainers have not had time to complete this manual process yet.

I'm actually working on streamlining that process right now. Once finished, it should result in a shorter response time to requests such as yours for index modification actions that aren't able to be handled by the automated system.