arduino/Arduino #6432: Confusion caused by the "Windows App" download option

per1234, 2017-06-27:
In the past week I've had two different forum users reply to my recommendation to upgrade to the latest version of the Arduino IDE with "I can't because it requires Windows 10 and I'm using an older Windows version". The problem is they are clicking the "Windows App" link on the [Software page](https://www.arduino.cc/en/Main/Software), then once they get to the Windows Store it tells then that Windows 10 is required. Encountering two people who have this mistaken impression over such a short time span makes me suspect this is going to be a fairly common issue.

I have a couple of ideas on how to make this sort of confusion less likely:
- Remove the "Get" icon next to the the "Windows app" link. I think this is drawing their eye. The download links don't obviously appear as button, they're just some text that's not even colored in the standard link blue but that icon just begs to be clicked. I see no value to that icon, it just causes visual clutter.
- Change the text of the "Windows app" link to something like "Windows app (requires Windows 10)" and add some text to the standard Windows download links that says something like "Windows XP and newer".

The most recent occurrence of this issue:
http://forum.arduino.cc/index.php?topic=485903.msg3317007#msg3317007

matthijskooijman, 2017-06-27:
Both suggestions look good to me.

facchinm, 2017-07-03:
Adding @00alis to the loop, thanks!

00alis, 2017-07-03:
I totally hear your point and I had similar concerns, unfortunately we have no freedom on how we display this badge, we need to follow Microsoft guidelines as reported here https://developer.microsoft.com/en-us/store/badges

So I don't think there is much we can do :(

matthijskooijman, 2017-07-03:
Then perhaps just leave out the badge entirely? Alternatively, the "Get it on Windows 10" badge could be useful, though it is quite large by default, it displays 104px high on that page:

[image]

Scaling down sorta works, but 26px high is already a bit too small I think:

[image]

40px might be ok?

[image]

00alis, 2017-07-04:
As said this is part of an agreement with Microsoft and there is not much we can do

kengdahl, 2017-12-27:
Well I did some reformating that is still within the bounds of the windows agreement. Take a look.

00alis, 2017-12-27:
@kengdahl where would I see the reformatting? As I said previously the brand guideline by Windows are pretty strict and there is no room for restyling...

kengdahl, 2018-01-02:
@00alis https://www.arduino.cc/en/Main/Software. Where it says "Windows app Requires Win 8.1 or 10"

matthijskooijman, 2018-01-02:
@kengdahl, perhaps it would be useful to add "For all versions of Windows" along with the other links? (or perhaps "Requires Windows xxx or newer, since there probably is a minimum windows version. Probably 7 or newer, IIRC XP does not reliably work anymore).

per1234, 2018-01-02:
I believe the Windows XP problem was fixed in https://github.com/arduino/Arduino/pull/5863. I haven't seen any problems reported relating to XP on the Arduino forum since the subsequent release.

kengdahl, 2018-02-13:
Can I just ask if the page, in its current state, is satisfactory now?

matthijskooijman, 2018-02-13:
It seems acceptable to me. It might still be unclear to people what version they need exactly (e.g. on Windows 10, is the store version actually better than the download version?), but it seems less confusing for Windows < 8 users to me. Also, I don't *really* care about Windows :-p

per1234, 2018-02-13:
The changes you've made go a long way towards preventing this issue. Thanks!

I do think it would be nice to state the supported Windows version range for the standard IDE to avoid any chance of someone thinking the version range text in the App link applies to the standard IDE as well.

kengdahl, 2018-03-22:
Do you guys know which versions of windows the current IDE works on?

facchinm, 2018-03-23:
Down to XP (for the normal IDE), Vista//7 for the Beta channel (due to `clang` incompatibilities with XP)

edwardvmills, 2018-11-10:
this download icon setup is a serious problem. i almost came to the conclusion that i needed to set up a linux machine to avoid the microsoft store. life is too short to try to do anything practical through anything 'app-like' or 'touch friendly'.

i am not a linux zealot. i run windows 10 for 95% of my home computer use. i've only played around with arduino through VNC/raspberrypi so far. i finally wanted arduino on my real desktop to look at the marlin firmware for a homebrew CNC. in this use case, running arduino from the windows desktop makes sense. i literally almost wrote off windows as a usable platform for arduino because of this tactic to channel me to an 'app store' against my will.

What's so wrong with an app-store? Well, for starters, i ended up unexpectedly on a microsoft domain after being DELIBERATELY MISLED BY A CALCULATELDLY DECEPTIVE ICON PLACEMENT. that was the first strike, and also the last strike.

i had to do multiple google searches to find the issue, since all the google results were polluted with pro-microsoft and pro-app store garbage. It is almost blind luck that i found this github thread, which made me realize i had gotten played by the microsoft store.....BEFORE I EVEN SET FOOT IN IT.