arduino/Arduino #6433: Wrong schematic (and text) in ShiftOut Tutorial

Liam18, 2017-06-27:
The capacitor is shown between LatchPin and GNS. However, it should be placed between VCC and GND.

See [this question on the Arduino StackExchange page](https://arduino.stackexchange.com/questions/39827/is-the-shiftout-tutorial-on-the-official-arduino-website-really-wrong-regarding) for more information.

It has been discussed in the forum as well: https://forum.arduino.cc/index.php?topic=408135.0

Thank you. And please tell me, if there's something I could do or if I did something wrong. 😳

Liam18, 2017-06-27:
Another source for the corrected capacitor placement: [SN74HC595 - Texas Instruments](http://www.ti.com/lit/ds/symlink/sn74hc595.pdf) (See Figure 5 under 10.2 Typical Application)

wsshambaugh, 2017-09-01:
Adding another reference to where this was an issue: https://arduino.stackexchange.com/questions/44291/weird-output-from-chained-74hc595-shift-registers

The shiftOut function has an operating frequency that is fast enough to be very sensitive to a non-decoupled power supply, and so getting the capacitor right here is more than a technicality - it's essential for the solid operation of the circuit. 0.1uF was failing in the above link, and 10uF was working. Recommend changing the drawing cap value to 10uF. 

Also, as brought up in the above posts, there is an overcurrent potential on the chips for LEDs with the resistor values as listed at 220ohm. Bumping them up to 470ohm should resolve this.

SimonePDA, 2018-04-02:
Fixed schematics with new 470 ohm resistors and 10uF capacitor wired to VCC and GND.

Thanks for pointing out these changes.

closed 2018-04-02 by SimonePDA