arduino/Arduino #6434: Arduino IDE   Save and Save As

Feroli13, 2017-06-28:
Hello,
Please Help.
On My arduino IDE, The option (File- Save)  and (File - Save As) are not active, so I Can't save my Sketck.
How to solve?
Thanks.

Fernando
[image]

Feroli13, 2017-06-28:
Hello,
Please Help.
On My arduino IDE, The option (File- Save) and (File - Save As) are not active, so I Can't save my Sketck.

[image]

How to solve?
Thanks.

Fernando

cmaglie, 2017-06-28:
You've the "Use external editor" preference set (that means that your sketch is already saved).

Feroli13, 2017-06-28:
OK.

Solved.

Thanks.

Fernando

closed 2017-06-28 by cmaglie