arduino/Arduino #6435: digitalRead(array[index]) doesn't work when array is in PROGMEM but works fine when it's in RAM

10gitu, 2017-06-28:
Environment:
  IDE Arduino 1.8.3 . 
  Board:Arduino Mega2560. 
  Just testing some hardware and playing with PROGMEM and F().

When using a regular or const array in RAM, digitalRead works fine. When moving it to PROGMEM it stops working for some reason. Presuming it's a compiler or library bug.

Here is a sample code with comments. (Note: this test code was done as part of employment so it might be copyrighted.)

#include <LiquidCrystal.h>

LiquidCrystal lcd(53,52,51,50,49,48,47);
unsigned long m1,m2; int c,ma,mi;
float av;
#define wnd 25 /*window for rolling average*/
char but[8]={0,0,0,0,0,0,0,0}; //last (current) state of buttons
const char bp[8]={30,31,32,33,34,35,36,37}; //pins; U D L R esc + ent -   ; not working when stored in PROGMEM
char deb[9]={0,0,0,0,0,0,0,0}; //debounce counter
#define dbc 50 /*debounce time in ms*/
char i,j,k;//small loop counters and tmp vars

void setup() {
#define pu(x) pinMode(x,INPUT_PULLUP);
  pu(30);pu(31);pu(32);pu(33);pu(34);pu(35);pu(36);pu(37);
  pinMode(13,OUTPUT);
  av=50;
  lcd.begin(20, 4);//cols,rows LCD2004 -> 20,4
  
  /*lcd.setCursor(0,0);  lcd.print("line0line0line0line0");  
  lcd.setCursor(0,1);  lcd.print("line1line1line1line1");
  lcd.setCursor(0,2);  lcd.print("line2line2line2line2");
  lcd.setCursor(0,3);  lcd.print("line3line3line3line3");*/
  
  //lcd.print("abcdefghijklmnopqrstABCDEFGHIJKLMNOPQRSTuvwxyz78901234567890!@#$%^&*()-=-=`~[]{}");
  //??? strange sometimes with line wrapping; tilde shows as right arrow on this lcd
  m1=0;ma=0;mi=1000;c=100; //skip first 500 cycles from stats
  k=0;

 Serial.begin(115200);

}

#define lp lcd.print  
// *.print functions don't take varargs in Arduino's libraries
#define lpf(x) lp(F(x))  /*string in flash, not ram*/
#define sp Serial.print
#define spf(x) sp(F(x)) /* flash */

void loop() { 
  //compute and display stats
  if (c) {--c;av=millis()/100;}//skip first n cycles
  
  ++k; digitalWrite(   13,   ( (k<10) || (k>20)&&(k<30) )   ); k=k%65; //blink pattern in loop cycles
  m2=millis();
  if (m1<=m2) {
     if (!c) { 
              if (ma<(m2-m1)) { ma=m2-m1;}
       if (mi>(m2-m1)) { mi=m2-m1;}
       lcd.setCursor(0, 3);  //col, line; zero-based
       lp (mi); lp (F("/")); lp (av) ; lp(F("/")); lp (ma); lp(F(" ")); lp(m2); lp(F("  "));
    }
    av=(av*wnd + m2-m1)/(wnd+1); //rolling window average
  }else {spf("millis overflow: m1="); sp (m1); spf(" m2=");Serial.println(m2);}
  m1=m2;
  
  lcd.setCursor(0,0);
  for (i=0;i<8;i++) { //poll and display buttons
    j=1-digitalRead(bp[i]); //this one needs the array in RAM not PROGMEM for some reason
    if (but[i]==j) { deb[i]=0;} else {++deb[i];} //debouncing
    if (  deb[i]>( dbc/( (int)av ) ) && deb[i]>3  ) {  //min 3 cycles or debounceMS/avgMS
       but[i]=(1-but[i]);
       if (Serial) {sp(F("Button "));sp(i,DEC);sp(F("="));sp(but[i],DEC);sp(F("\n"));}
    }
    lp(F("B"));lp((int) i);lp(F("="));lp( (int) but[i]);lp(F(" "));
    if (i==3) lcd.setCursor(0,1); //buttons 4..7 come on 2nd line of the lcd, otherwise it may wrap in funny ways
 }//buttons
}

cmaglie, 2017-06-28:
> When moving it to PROGMEM it stops working for some reason

This is an expected behaviour for AVR processors.

BTW this is not the place to ask question about sketchs, please search on google or discuss your sketch in the forum for a solution:

https://forum.arduino.cc/

closed 2017-06-28 by cmaglie

10gitu, 2017-06-28:
I have the solution - keeping the pin list in RAM. No biggie, since the mega doesn't have that many pins anyway. (Or one can use macros etc.)
Lesson learned: if the need to optimize ever comes, do it one by one.

The issue was that somehow the compiler (not the CPU) is unable to issue instructions to pull data from flash (e.g. by using a tmp 1-byte variable), nor does it issue a warning when unable to do so, especially since it's a "known issue" (I would not call it "expected" - The expectations are for it to either work or show a compile-time error.)

cmaglie, 2017-06-28:
> The expectations are for it to either work or show a compile-time error

The compiler can show an error only when there is a **syntax error** and your sketch has **no** syntax errors.

Anyway the absence of syntax errors doesn't mean that the sketch logic is correct and, in your case, you used PROGMEM in the wrong way (you must use the `pgm_read_xxxx` functions from `avr/progmem.h`) hence my suggestion to google or ask the forum.