arduino/Arduino #6437: Can't use SoftwareSerial library inside a custom library.

whileloop123, 2017-06-28:
I'm trying to get SoftwareSerial library to work in a custom library that I'm working on, but it seems that the code execution gets stopped somewhere and the void loop() function never executes.
I know that SoftwareSerial is causing the issue because when I comment out begin() in the constructor, the loop function executes.

Please be easy on me. I'm a newbie.

Thank you for your help.

MobileBLE.h
```C++
#ifndef _MobileBLE_h
#define _MobileBLE_h

class SoftwareSerial;

class MobileBLE
{
	public:
		MobileBLE(int TX, int RX);
		char getButton();
		int getThrottle();
		int getSteering();
		int checkBluetooth();
	private:
		SoftwareSerial &_bluetooth;
		int _signal[10];
		int _signalLength = 0;
		int _TX;
		int _RX;
		int _throttle;
		int _steering;
		char _button;
		void begin();
		void push(int elem);
		void deleteElements();
};

#endif
```

MobileBLE.cpp
```C++
#include "MobileBLE.h"
#include "Arduino.h"
#include <SoftwareSerial.h>

MobileBLE::MobileBLE(int TX, int RX) {
	_TX = TX;
	_RX = RX;
	_signalLength = 0;
	begin();
}

void MobileBLE::push(int elem) {
	_signal[_signalLength] = elem;
	_signalLength++;
}

void MobileBLE::deleteElements() {
	for (int i = 0; i < _signalLength; i++) {
		_signal[i] = 0;
	}
	_signalLength = 0;
}

void MobileBLE::begin() {
	_bluetooth = SoftwareSerial(_TX, _RX);
	_bluetooth.begin(115200);
	delay(100);
	_bluetooth.println("U,9600,N");
	_bluetooth.write("AT+ROLE1");
}

int MobileBLE::checkBluetooth() {
	// returns 0 - if no transmission found for throttle, steering, or button
	// returns 1 - if transmission found for throttle and steering
	// returns 2 - if transmission found for button

	for (int i = 0; i < 3; i++) {
		if (_bluetooth.available()) {
			int input = _bluetooth.read();

			push(input);

			if (input == 254) {
				// end of transmission for throttle and steering values
				_throttle = _signal[0];
				_steering = _signal[1];
				deleteElements();

				return 1;
			}
			else if (input == 255) {
				// end of transmission for button click
				_button = (char)_signal[0];
				deleteElements();

				return 2;
			}
		}
	}

	return 0;
}

char MobileBLE::getButton() {
	//return _button;
	return 'a';
}

int MobileBLE::getThrottle() {
	return _throttle;
}

int MobileBLE::getSteering() {
	return _steering;
}
```

per1234, 2017-06-29:
Also posted at:
http://forum.arduino.cc/index.php?topic=486354

matthijskooijman, 2017-06-29:
This is a question for the forum, since it is likely something in your code, so I'm closing this issue. If you dig further and find this is a bug in SoftwareSerial or other parts of Arduino, feel free to leave a comment here or open another issue.

closed 2017-06-29 by matthijskooijman